gqrx-sdr/gqrx #485: Support for decoding DAB

bogdanr, 2017-02-10:
Hello,

It would be really nice if GQRX could decode DAB signals.
Any chance to implement it?

csete, 2017-02-11:
Hi @bogdanr 

In 2017 I am focusing my efforts on new "lighter" SDR applications that also have transmit capabilities and that will not leave much time for gqrx, sorry.

kantooon, 2017-02-11:
Hi Bogdan,
I looked into a DAB plugin, but DAB is a complex beast. I don't think the pain/gain is worth it, unless you live in Norway.
Numai bine,
Adi

On 10 February 2017 15:27:22 GMT, "Bogdan Rădulescu" <notifications@github.com> wrote:
>Hello,
>
>It would be really nice if GQRX could decode DAB signals.
>Any chance to implement it?

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

csete, 2017-02-11:
That's pretty much also my philosophy... I really enjoy DAB but it is possible to buy a good DAB radio for the fraction of the price it would cost to implement it in SDR, so I'd rather spend my time on working on SDR transmitters.

By the way, I think there is a java sdr which has DAB support, so perhaps porting it to C/C++/gnuradio could be done for little effort.

kantooon, 2017-02-11:
There is a gnuradio OOT module, but it only demodulates. I tried it, but I would not implement decoding for free, it's too complicated.
Cheers,
Adi

On 11 February 2017 12:56:25 GMT, Alexandru Csete <notifications@github.com> wrote:
> That's pretty much also my philosophy... I really enjoy DAB but it is
> possible to buy a good DAB radio for the fraction of the price it would
> cost to implement it in SDR, so I'd rather spend my time [...]

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

bogdanr, 2017-02-12:
Thanks all for having a look at this.
Just as a reference, there is a starting point on [wiki.opendigitalradio.org](http://wiki.opendigitalradio.org/DAB_reception).

@csete Looking forward to see what you have planed for 2017. Transmitting would be much nicer than DAB anyways.

@kantooon If you are from Bucharest maybe we could get in touch and share some heatmaps and experiences. My contact details are [HERE](https://bogdan.nimblex.net/cv/)

andimik, 2017-02-14:
For DAB please use qt-dab or welle.io (both here in Github.com) supporting DAB sticks, SDRPLAY and Airspy and rtltcp.

csete, 2017-03-19:
https://www.welle.io/
https://github.com/AlbrechtL/welle.io

closed 2017-03-19 by csete

roelandjansen, 2017-10-19:
https://www.sdr-j.tk/index.html