gqrx-sdr/gqrx #489: Fails to load LimeSDR backend every second time

csete, 2017-02-19:
Every second time gqrx is started with LimeSDR, the soapysdr backend fails to connect to the device and prints:

```
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00461618'

########################################################
##   !!!  Warning: gateware version mismatch  !!!
## Expected gateware version 2, revision 6
## But found version 0, revision 0
## Follow the FW and FPGA upgrade instructions:
## http://wiki.myriadrf.org/Lime_Suite#Flashing_images
## Or run update on the command line: LimeUtil --update
########################################################

...

FATAL: SoapyLMS7::setSampleRate() -- rate too high

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

Failed to set RX input rate to 1e+07
Your device may not be working properly.

libusb: warning [libusb_exit] application left some devices open
```

guruofquality, 2017-03-05:
I ctrl+c apps a lot, and this hasnt happened to me FWIW. Just curious: Can we confirm that the gr-osmosdr blocks are being destructed properly when gqrx is shutdown? I don't like the idea that an unexpected shutdown would make the limesdr unrecoverable, and that in itself is a problem. But we should also be doing nice shutdowns when possible.

csete, 2017-03-05:
Hehe, I'm not sure one properly destructs GNU Radio blocks in C++. As I recall I was told once that I am not suppose "delete ..." GNU Radio blocks.