gqrx-sdr/gqrx #490: LimeSDR bandwidths

csete, 2017-02-19:
- 0 MHz selects auto
- 3.6 MHz and up can be used for manual setting
- 60 MHz is used for HF below 30 MHz
- Below 3.6 MHz throws an exception and gqrx crashes.

cwshep, 2020-02-03:
I think I can verify a related issue with uncaught exceptions crashing gqrx on Iris (a Lime transceiver using SoapySDR).

Notably, if the LNB LO is not set or set below ~65 MHz on the "Configure I/O devices" dialog, then I get an immediate crash:

```
[ERROR] SetFrequencySXR(-25.764 MHz) - required VCO frequency is out of range [3800-7714] MHz
[ERROR] [RF3E000040] Iris::setFrequency(Rx, 0, -25.764 MHz) tune Failed - SetFrequencySXR(-25.764 MHz) - required VCO frequency is out of range [3800-7714] MHz
```

[image]

Once I open the app successfully, some frequencies will throw an exception that is uncaught by gqrx and crash it  (typically low frequencies that just happen while I'm typing it in, but it seems to tune in realtime rather than waiting for the entire input):

```
[ERROR] SetFrequencySXR(20 MHz) - required VCO frequency is out of range [3800-7714] MHz
[ERROR] [RF3E000040] Iris::setFrequency(Rx, 0, 20 MHz) tune Failed - SetFrequencySXR(20 MHz) - required VCO frequency is out of range [3800-7714] MHz
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

[INFO] Power down and cleanup
```

[image]

This is using code that was just pulled and built today, here's the startup info:

```~$ gqrx -r
gr-osmosdr 0.1.5 (0.1.5) gnuradio 3.7.11
built-in source types: file fcd rtl_tcp rfspace soapy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/asdf/.config/gqrx/bookmarks.csv
gr-osmosdr 0.1.5 (0.1.5) gnuradio 3.7.11
built-in source types: file fcd rtl_tcp rfspace soapy redpitaya 
[INFO] Initializing IRIS-030E - RF3E000040...
[INFO] LMS7002M rev 0x1, ver 0x7, mask 0x1
[INFO] Opening /etc/iris/calibration.sqlite3
[INFO] Initialization complete
[INFO] Using local ethernet interface: enp3s0
[INFO] mtu 1500 bytes -> 473 samples X 1 channels, 3 bytes per element
[INFO] Rx route: source[04] -> 0x6f0190[04]
```

argilo, 2020-11-28:
From the issue description I'm not quite sure what the issue was, but the crash part was fixed in #835 so I'll close this off. Let me know if there's still more to be done.

closed 2020-11-28 by argilo