gqrx-sdr/gqrx #491: pulseaudio related crash when running through X11 forwarding

bogdanr, 2017-02-26:
I am running gqrx through a SSH tunnel like this:
`ssh -Y -C 192.168.1.120 gqrx`

When I press the `Start DSP` button it crashes with the error that can be seen on the last line:

```
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9.2                                                                                            
built-in source types: file fcd rtl rtl_tcp hackrf rfspace                                                                           
../src/pulseaudio/pa_sink.cc: pa_simple_new() failed: Connection refused
Using Volk machine: sse4_1_64_mmx_orc
BookmarksFile is /root/.config/gqrx/bookmarks.csv
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9.2
built-in source types: file fcd rtl rtl_tcp hackrf rfspace 
Using device #2 Generic RTL2832U SN: 77771111153705700
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Assertion 'p' failed at pulse/simple.c:273, function pa_simple_write(). Aborting.
```

Of course, running gqrx locally works OK. The problem is when I run it over SSH. Firefox on the other hand runs OK over SSH and also I can hear sound on the remote machine.

csete, 2017-02-26:
> ../src/pulseaudio/pa_sink.cc: pa_simple_new() failed: Connection refused

Perhaps you don't have pulseaudio running on the computer?

bogdanr, 2017-02-26:
pulseaudio is running on both computers

csete, 2017-02-26:
Still, something needs to be configured. Sorry, but you'll have to figure this one out for yourself if you want it resolved quickly, because I don't have experience with this nor do I have time to research it.

Try searching for "pulseaudio ssh connection refused" - I'm sure others have had similar problems.

Note that firefox could be using alsa directly.

csete, 2017-02-26:
> Note that firefox could be using alsa directly.

Indeed, on Xubuntu the firefox package depends on libasound2 so it is not using pulseaudio. If you can't find any solution you could rebuild gqrx with portaudio support.

csete, 2017-04-09:
Any updates here. It seems to be a usage/configuration issue and not a gqrx issue.

fbraghiroli, 2017-04-09:
I had the same "problem" and I resolved following this guide:
http://psung.blogspot.it/2011/09/network-audio-with-pulseaudio-made.html

On paragraph "Piggybacking on X11 forwarding" it is explained how to forward the pulseaudio data from the remote source to the client. This procedure is valid for all the pulseaudio based applications, however it would be great to include these info in the gqrx documentation

csete, 2017-04-09:
@fbraghiroli - thanks for the info!

The wiki here on github is enabled and open to everybody to edit, so feel free to add a page with this note.

closed 2017-04-09 by csete

LinuxUser404, 2018-07-05:
I had a somewhat similar problem. I was testing sound using pulse server. It was working just as expected through ssh until I added '-X' or '-Y' option. The solution to the problem was setting PULSE_SERVER environment variable to IP of the computer with speakers(which could be the host you are connecting to).
`export PULSE_SERVER=192.168.10.200`
Same problem arises if you use vglconnect instead of "ssh -X". And the same solution solves it.