gqrx-sdr/gqrx #492: OS X Binary links to 2.6 build, not 2.6.1

luserppc, 2017-02-27:
The disk image you download is labeled "Gqrx-2.6-1.dmg", but the application is the previous build, 2.6.

csete, 2017-02-27:
Hi,

The label is correct. It means gqrx version 2.6 package version 1. There is no Mac OS X package of gqrx 2.6.1. If there will be it will be called Gqrx-2.6.1-1.dmg.

closed 2017-02-27 by csete

luserppc, 2017-02-27:
My apologies,  I see now what you mean. 
Csete. are there instructions on compiling GQRX from source on OSX via Xcode?  I see many tutorials online that use Homebrew, or ports,  but none on building natively w/ Xcode.
-Thanks.

csete, 2017-02-27:
I don't think it is possible, or lets say feasible, to build without using either macports or homebrew. Since gqrx is a "ported" software it has many dependencies not readily available on Mac OS X. So if you start building and installing everything one by one, you will end up recreating the environment provided by macports or homebrew.

luserppc, 2017-02-27:
I see.  Well,  regardless, it's wicked cool that I can run it on OSX at all.  It's a great program, thank you for taking the time to answer my questions.

csete, 2017-02-28:
You are welcome.