gqrx-sdr/gqrx #493: .conf files are deleted when changes are saved.

romeojulietthotel, 2017-02-28:
Using git master.

Steps to reproduce:

1. make some changes and save them to atest.conf
2. ls -l ~/.conf/gqrx
3. make some changes and save them to atest.conf
4. when asked to overwrite answer yes
5. ls -l ~/.conf/gqrx

If I can figure it out I will send a PR.

alexf91, 2017-02-28:
I can reproduce this.
The debug output for the second save:
```
Saving configuration to: "/home/alex/.config/gqrx/atest.conf"
File "/home/alex/.config/gqrx/atest.conf" already exists => DELETING...
Deleted "/home/alex/.config/gqrx/atest.conf"
Error saving configuration to "/home/alex/.config/gqrx/atest.conf"
```

Relevant code is in mainwindow.cpp:saveConfig.
When newfile = oldfile, then the copy operation fails.

csete, 2017-02-28:
I assume we are talking about using the "Save as..." funcionality and now the usual auto-save when the program exits.

alexf91, 2017-03-01:
@csete That's right.

It would be enough to check if newfile == oldfile and don't do anything (just sync) if they are equal to resolve this bug and keep the current behavior.
The current "Save as" behavior seems counterintuitive to me, because all settings are written to the current configuration before saving them to another file. Except for the 'crashed' field, all settings will be equal.

romeojulietthotel, 2017-03-01:
We should use:
http://doc.qt.io/qt-5/qsavefile.html

alexf91, 2017-03-02:
@romeojulietthotel I don't think this makes sense here. QSaveFile is for raw file IO. QSettings already uses QSaveFile internally when writing to disk.

romeojulietthotel, 2017-03-02:
QSaveFile makes sense to me.
`The QSaveFile class provides an interface for safely writing to files.
`
Using copy is very clearly wrong.
http://doc.qt.io/qt-5/qfile.html#copy

`Note that if a file with the name newName already exists, copy() returns false (i.e. QFile will not overwrite it).
`

alexf91, 2017-03-02:
What QSaveFile does is writing to a temporary file, and then moving it to the desired location when commit() is called. It avoids incompletely written files (e.g. when the application crashes during writing).

In our case, the call to copy failed because the source file was deleted before calling copy. I don't even know how we would use QSaveFile here, other than implementing our own copy operation.
I don't see any advantage in doing that.

romeojulietthotel, 2017-03-02:
Copy will fail as it is used even if the file is not deleted.

http://doc.qt.io/qt-5/qtwidgets-mainwindows-application-mainwindow-cpp.html

alexf91, 2017-03-02:
Copy won't fail if newfile doesn't exist.
I see three cases:
* newfile == oldfile: Triggers the error described
* newfile != oldfile && newfile doesn't exist: No delete, oldfile copied to newfile
* newfile != oldfile && newfile exists: delete newfile, oldfile copied to newfile

The copy operation only fails in the first case, because the source file is deleted. This is addressed in #495.
I don't see a scenario where the copy operation fails due to an existing destination file.
Am I missing something?

csete, 2017-03-05:
@romeojulietthotel  can you please check and confirm that the applied fix solves your issue? Thanks.

PR #495 (2017-03-05): Check if new and old configuration file are equal
Fixes #493 
Keeps the old behavior of syncing before "save as", but checks if old and new configuration files are equal.
closed 2017-03-05 by csete