gqrx-sdr/gqrx #494: airspy-mini: choppy audio from FM demodulators

kzyapkov, 2017-03-01:
All FM demodulators - Narrow FM and WFM mono and stereo - produce choppy audio output with an AirSpy Mini; both 3MHz or 6MHz sample rates. This is on an up-to-date Arch Linux, [gqrx-git](https://aur.archlinux.org/packages/gqrx-git/) version v2.6-73-g80f0.

Some debug info:

```
$ airspy_info 
airspy_lib_version: 1.0.9

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-0-g946184a 2016-09-19
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x04A464C837XXXXXX
Supported sample rates:
	6.000000 MSPS
	3.000000 MSPS
Close board 1

$ gqrx 
linux; GNU C++ version 6.2.1 20160830; Boost_106300; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Resampling audio 96000 -> 48000
BookmarksFile is /home/kiril/.config/gqrx/bookmarks.csv
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using AIRSPY with firmware AirSpy MINI v1.0.0-rc10-0-g946184a 2016 

```

csete, 2017-03-01:
You need a faster computer, run volk_profile to enable SIMD in gnuradio, build gqrx with -DCUSTOM_AIRSPY_KERNELS=on and use decimation, or .... 

Also be sure that everything is compiled as release build and not some unoptimized debug build.

The reason why it only happens with FM is because FM requires more CPU than other modes.

The reason why it only happens with airspy and not rtl dongles is that airspy uses the driver to convert samples and that requires more CPU (you can check it with airspy_rx using complex float sample type vs rtl_test).

I'm closing this since it is not a bug but a rather well known fact.

closed 2017-03-01 by csete

kzyapkov, 2017-03-02:
Actually, my problem was identical to this guy's:

https://groups.google.com/forum/#!topic/gqrx/9AI0Rcf3kSE

and got fixed in the same manner. I suppose the sample rate mismatch manifested as the high-pitched, choppy audio I was hearing.

csete, 2017-03-02:
Ah, ok, it makes sense. Unfortunately, gqrx can not see whether it is an Airspy Mini or R2, so we have to list all sample rates.