gqrx-sdr/gqrx #496: Frequency calibration crash  gqrx/Pothos Odroid-C2/Ubuntu16

KC7NOA, 2017-03-12:
When you enter a value into the ppm box gqrx just closes ...
All built from Git

I was highlighting 0 and hit any number...

what was seen in terminal

INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 4.4057 MHz)
SetFrequency using cache values vco:0, csw:13
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, -25.5943 MHz)
[INFO] SoapyLMS7::setFrequency(Rx, 0, CORR, 1e-06 MHz)
terminate called after throwing an instance of 'std::runtime_error'
  what():  SoapyLMS7::setFrequency(CORR) unknown name
Aborted

csete, 2017-03-12:
Thanks for the report, but this is IMO a problem in the backend or the gr-osmosdr interface. Basically, gqrx is just a GUI built on top of a hardware abstraction layer, so if a hardware backend does not support a setting it should just ignore it.

Also, I am not sure why you always mention Pothos when you talk about gqrx... They are two entirely different applications.

KC7NOA, 2017-03-13:
I cant tell from the terminal -- where the output came from -- Pothos/SoapySDR or gqrx ....
I say Pothos because SoapySDR has Pothos as the parent.

https://github.com/pothosware/SoapySDR

csete, 2017-03-13:
No problem, it is quite confusing most of the time :)

Pothosware is the company/organisation who makes SoapySDR and Pothos.

KC7NOA, 2017-04-16:
I also have an issue with another computer .. that gqrx does not find soapy as a source ....

csete, 2017-04-16:
Ok, but this issue tracker is intended to track issues in gqrx, not issues that people are having with their computers. It's best to post such issues on a user forum where other users can help.

KC7NOA, 2017-04-17:
Im not sure if its a problem with gqrx or soapy or gnuradio or lime ... the computer is fine.

Soapy detects the lime and i can probe it ... so im sure its not LimeSuite/driver & Soapy

csete, 2017-04-23:
I see Josh has fixed this in gr-osmosdr:
http://cgit.osmocom.org/gr-osmosdr/commit/?id=a9e536f45bae3c283a1ad7ed13ff17bbda6173ba

The fix will be available in the next packages.

closed 2017-04-23 by csete