gqrx-sdr/gqrx #497: GQRX doesn't listen for remote control connections if $http_proxy is set.

kovinesinsila, 2017-03-13:
It's a weird one. I'm using GQRX on linux and it doesn't listen on control socket if launched from an environment that has http_proxy set.

alexf91, 2017-03-14:
Interesting. Qt determines the proxy settings automatically on application level when no proxy explicit proxy setting is specified.
http://doc.qt.io/qt-5/qnetworkproxyfactory.html

Should be a one line fix.

csete, 2017-03-14:
I do not understand what the problem is. Can you please provide some more info about how to reproduce it and also how you determine that gqrx is not listening?

alexf91, 2017-03-14:
If http_proxy is set, "lsof -i" doesn't show that gqrx is listening on a socket, and any connection attempt is refused. If it is not set, I get
```gqrx      28263 alex   18u  IPv6 14742039      0t0  TCP *:7356 (LISTEN)```

I don't have a proxy running, so I just set http_proxy to some value. This might be a problem here. I'm in no way a network expert, but I think that gqrx should keep listening for local connections even if a proxy is set.

I found this old bug report for Qt, but there's not much information:
https://bugreports.qt.io/browse/QTBUG-48194

csete, 2017-03-14:
Thanks for the info @alexf91

alexf91, 2017-03-14:
It's a bug in Qt.

Here is a similar bugreport for UDP: https://bugreports.qt.io/browse/QTBUG-58374
And here's the corresponding fix: https://codereview.qt-project.org/#/c/186124/

If I check the return value of QTcpServer.listen(), it fails when http_proxy is set.

csete, 2017-03-14:
@kovinesinsila 
Can you check whether the applied patch fixes your problem?

PR #499 (2017-03-14): Disable detected proxy settings
Should fix #497 
Not sure if this should be merged, because it disables the proxy detection used by Qt. I'm not even entirely sure why this caused problems in the first place.
closed 2017-03-14 by csete