gqrx-sdr/gqrx #498: Buffer fill up FFT latency

KC7NOA, 2017-03-14:
Im not sure where this issue is located ...

I know that when set to FM and ran for a while that the latency for the FFT and audio playback gets real long ... almost 10 seconds delayed (more?)

Quickest way to get back to real time is to simply stop gqrx and start again.

Sorry if this has already been expressed .... im not sure exactly how to discribe it.

Odroid-C2/Ubuntu16 , all built from git

csete, 2017-03-14:
It could be that the odroid can not keep up at that sample rate and one of the blocks (probably FM demod) runs slower and builds up the latency. You can check using a tool like htop - it should show 1 core running at close to 100%.

Please note that the preferred channel for user support is the mailing list, see http://gqrx.dk/user-support, and we only use github as a developer tool.

csete, 2017-03-14:
By the way, did you run the volk_profile tool?

KC7NOA, 2017-03-14:
Im not 100% sure i did on the C2 ... I have before ...

And it will be a easily be a few days before i can check .. i have taken the setup apart to try to find a suitable case for the Odroid and LimeSDR. That i dont need to cut a bunch of holes into.

csete, 2017-03-14:
No problem, just post when you have update.

csete, 2017-04-09:
Any updates here? If not, we should close this.

closed 2017-04-15 by csete