gqrx-sdr/gqrx #506: GQRX standard interface to another GNU Radio flowgraph

mhostetter, 2017-03-31:
This is a feature request.

Maybe this has already been thought of, but here's what I'm thinking. It would be nice to have a standard interface from GQRX to another GNU Radio flowgraph. The goal I have in mind is to use GQRX to control the SDR and for spectrum display (because it's just better than anything native in GNU Radio), but pass the IQ samples to another flowgraph for custom GNU Radio plug-and-play.

The interface could be something like UDP. Maybe there could be an additional "Mode" that is something like "GNU Radio". And then there could be a `gr-gqrx` OOT module that has the companion GQRX source block.The benefit I see in this approach is that GQRX could pass the center frequency and sample rate to the flowgraph.

Maybe in the "GNU Radio" mode the user could fix the bandwidth/sample rate, but would still be able to pan around and adjust the tuned frequency.

csete, 2017-03-31:
We already have a TCP interface for control and a UDP interface for audio steaming:
http://gqrx.dk/doc/remote-control
http://gqrx.dk/doc/streaming-audio-over-udp

The audio streaming is currently fixed at 48 kHz and one channel, but I'd rather see that improved than adding yet another interface.

mhostetter, 2017-03-31:
Ok.

Ideally for my application, I'd like to stream complex IQ at a settable sample rate (maybe even several Msps), centered at the frequency cursor. Then I could just have a flowgraph with a UDP source followed by my custom DSP and plotting.

Again, ideally :smile:

csete, 2017-03-31:
The reason I pointed out the UDP/audio interface is because it is already very close to what you want, just with some limitations. Don't let the term "audio" fool you. Gqrx has a "Raw I/Q" mode where I/Q samples are filtered, resampled and then sent directly to the audio output. Needs some work, but it's very close.

In any case, somebody else will have to work on this because I work on other projects at the moment.

mhostetter, 2017-03-31:
@csete well if you wouldn't mind pointing me in the right direction (source files/functions to start with), I can fork the repo and take a crack at it. Thanks in advance.

alexf91, 2017-03-31:
If we can set the sample rate of the audio up to the RF sample rate, we would be pretty close to the what you want to achieve.

@mhostetter Relevant files are:
* src/applications/gqrx/receiver.{cpp,h} for the flowgraph
* src/applications/gqrx/remote_control.{cpp,h} for the TCP interface
* src/qtgui/dockaudio.{cpp,h,ui} for the audio dockwidget

I would split the whole process in a few steps:
* Add option to set the audio sample rate to the receiver and the dockwidget
* Add option to set the audio sample rate to the remote control interface
* Rework the flowgraph, widgets and remote control to allow different settings for the audio sample rate and the UDP sample rate

The third step also implies splitting the audio and UDP path in some cases. because Pulseaudio won't be happy with 20 Msps.

Somewhere along the line, we could also implement #341.

argilo, 2020-11-28:
I'll close this off, since stereo UDP streaming was implemented in #748 and there's an open issue for higher sample rates in #868.

closed 2020-11-28 by argilo