gqrx-sdr/gqrx #507: Gqrx hangs when NFM deemphasis settings are changed after recording

alexf91, 2017-04-03:
Changing the NFM settings causes Gqrx to hang in some cases after audio was recorded:
* Start and stop audio recording
* Change NFM Tau from "Off" to something else or from some value to "Off"
* Watch Gqrx hang

Here's a stacktrace after sending SIGINT:
```
#0  0x00007ffff7105c00 in gr::hier_block2::lock() () from /usr/lib/libgnuradio-runtime-3.7.10.1.so.0.0.0
#1  0x00000000005d8b36 in rx_demod_fm::set_tau (this=0xabcae0, tau=0)
    at /home/alex/projects/gqrx/src/dsp/rx_demod_fm.cpp:152
#2  0x000000000065b581 in nbrx::set_fm_deemph (this=0xb3d620, tau=0)
    at /home/alex/projects/gqrx/src/receivers/nbrx.cpp:266
#3  0x00000000004fc85a in receiver::set_fm_deemph (this=0xb3a140, tau=0)
    at /home/alex/projects/gqrx/src/applications/gqrx/receiver.cpp:923
#4  0x00000000004e984f in MainWindow::setFmEmph (this=0x7fffffffda30, tau=0)
    at /home/alex/projects/gqrx/src/applications/gqrx/mainwindow.cpp:1129
#5  0x0000000000662eea in MainWindow::qt_static_metacall (_o=0x7fffffffda30, _c=QMetaObject::InvokeMetaMethod, 
    _id=15, _a=0x7fffffffc530)
    at /home/alex/projects/gqrx/build/src/gqrx_automoc.dir/moc_mainwindow_NB6I6GI4AJDIMH.cpp:406
#6  0x00007ffff3c81d49 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/libQt5Core.so.5
#7  0x000000000066ae7c in DockRxOpt::fmEmphSelected (this=0xbd9320, _t1=0)
    at /home/alex/projects/gqrx/build/src/gqrx_automoc.dir/moc_dockrxopt_LVQZ3TKUDSVXAZ.cpp:474
#8  0x00000000006297ea in DockRxOpt::demodOpt_fmEmphSelected (this=0xbd9320, tau=0)
    at /home/alex/projects/gqrx/src/qtgui/dockrxopt.cpp:668
#9  0x000000000066a3d9 in DockRxOpt::qt_static_metacall (_o=0xbd9320, _c=QMetaObject::InvokeMetaMethod, _id=34, 
    _a=0x7fffffffc8c0) at /home/alex/projects/gqrx/build/src/gqrx_automoc.dir/moc_dockrxopt_LVQZ3TKUDSVXAZ.cpp:278
```

csete, 2017-04-03:
I am aware of this but it looks like error occurs in gr::hier_block2::lock() so I don't know what we can do in gqrx. Maybe a different version of boost would help.

vladisslav2011, 2022-01-25:
This bug is duplicate of https://github.com/gqrx-sdr/gqrx/issues/1030.
I think, we can close it, as workaround is merged and works well.

argilo, 2022-01-29:
Agreed, this does appear to be a duplicate of #1030. Closing.

closed 2022-01-29 by argilo