gqrx-sdr/gqrx #508: PyBOMBS.install_manager - ERROR - Error installing package gqrx. Aborting.

wdx7266, 2017-04-07:
I meet a problem, what should I do?

> # pybombs install gqrx
> PyBOMBS - INFO - PyBOMBS Version 2.3.0
> PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
> Install tree:
> |
> \- gqrx
> PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
> PyBOMBS.install_manager - INFO - Installing package: gqrx
> PyBOMBS.Fetcher - INFO - Already fetched: gqrx
> PyBOMBS.Packager.source - WARNING - Build dir already exists: /home/wdx7266/gnuradio-prefix/src/gqrx/build
> Configuring: (100%) [======================================================================================================================================================================================]
> Building:    (100%) [======================================================================================================================================================================================]
> [  1%] Automoc for target gqrx
> [  1%] Built target gqrx_automoc
> [  2%] Building CXX object src/CMakeFiles/gqrx.dir/dsp/filter/fir_decim.cpp.o
> /home/wdx7266/gnuradio-prefix/src/gqrx/src/dsp/filter/fir_decim.cpp: In constructor ‘fir_decim_cc::fir_decim_cc(unsigned int)’:
> /home/wdx7266/gnuradio-prefix/src/gqrx/src/dsp/filter/fir_decim.cpp:116:9: warning: ‘auto’ changes meaning in C++11; please remove it [-Wc++0x-compat]
>          auto stage = &decimation_stages[index];
>          ^
> /home/wdx7266/gnuradio-prefix/src/gqrx/src/dsp/filter/fir_decim.cpp:116:14: error: ‘stage’ does not name a type
>          auto stage = &decimation_stages[index];
>               ^
> /home/wdx7266/gnuradio-prefix/src/gqrx/src/dsp/filter/fir_decim.cpp:118:21: error: ‘stage’ was not declared in this scope
>          if (decim % stage->decimation == 0)
>                      ^
> make[2]: *** [src/CMakeFiles/gqrx.dir/dsp/filter/fir_decim.cpp.o] Error 1
> make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
> make: *** [all] Error 2
> PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
> PyBOMBS.Packager.source - ERROR - Problem occurred while building package gqrx:
> Build failed.
> PyBOMBS.install_manager - ERROR - Error installing package gqrx. Aborting.
>

csete, 2017-04-07:
Thanks for the heads up.

You can try now with the latest code. It should be fixed.

wdx7266, 2017-04-08:
Thank you very much! I successful install `gqrx` with pybombs.

closed 2017-04-08 by csete

vampire2310, 2019-09-22:
PyBOMBS.install_manager - INFO - Installing package: qwt6
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Building:    (100%) [=========================================================]
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /usr/local/src/qwt6/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/usr/local/src/qwt6/src'
compiling qwt_abstract_scale_draw.cpp
In file included from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qlocale.h:43,
                 from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qtextstream.h:46,
                 from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qdebug.h:49,
                 from qwt_interval.h:17,
                 from qwt_scale_div.h:14,
                 from qwt_abstract_scale_draw.h:14,
                 from qwt_abstract_scale_draw.cpp:10:
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h: In constructor ‘QVariant::QVariant(QVariant&&)’:
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:273:25: warning: implicitly-declared ‘constexpr QVariant::Private& QVariant::Private::operator=(const QVariant::Private&)’ is deprecated [-Wdeprecated-copy]
  273 |     { other.d = Private(); }
      |                         ^
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:399:16: note: because ‘QVariant::Private’ has user-provided ‘QVariant::Private::Private(const QVariant::Private&)’
  399 |         inline Private(const Private &other) Q_DECL_NOTHROW
      |                ^~~~~~~
In file included from qwt_scale_map.h:14,
                 from qwt_abstract_scale_draw.cpp:14:
qwt_transform.h: At global scope:
qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type
  110 |     QT_STATIC_CONST double LogMin;
      |     ^~~~~~~~~~~~~~~
qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type
  111 |     QT_STATIC_CONST double LogMax;
      |     ^~~~~~~~~~~~~~~
make[1]: *** [Makefile:3819: obj/qwt_abstract_scale_draw.o] Error 1
make[1]: Leaving directory '/usr/local/src/qwt6/src'
make: *** [Makefile:48: sub-src-make_first-ordered] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package qwt6:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package qwt6. Aborting.

vampire2310, 2019-09-22:
need help