gqrx-sdr/gqrx #511: Error on compiling using cmake

projectbugar, 2017-04-17:
Hi, honestly I'm new to SDR platforms. As I  compile gqrx on rpi3(running on raspbian jessie) I got this error:

CMake Error at CMakeLists.txt:101 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/gqrx.git/build/CMakeFiles/CMakeOutput.log".

// Ive tried to find Qt5Config.cmake and qt5-config.cmake  but did not found it. Is there anything that I'm missing to install? I really need help! Thanks!

csete, 2017-04-17:
Hi,

Sounds like you have no idea what you are doing, so perhaps you should stick to the binaries available on the website: http://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi

If you insist on building from source, please start with a background search.

If you need further help, please follow the instructions on http://gqrx.dk/user-support

closed 2017-04-17 by csete

LongnoseRob, 2017-04-17:
GQRX requires QT5, so you should check if you have all needed packages installed as mentioned here:
https://github.com/csete/gqrx#installation-from-source

The better way to find help for this is the mailing list mentioned also here:
https://github.com/csete/gqrx#getting-help-and-reporting-bugs

projectbugar, 2017-04-17:
@csete 
Ive just tried using the binaries youve mentioned and got no luck, gqrx cant be open, it crashes.