gqrx-sdr/gqrx #512: Tooltips are black on black in the mainwindow

chron-isch, 2017-04-22:
Hi,

thanks for all the work, I love the program!

It's not really a bug and shouldn't cause you much work, but it's one of my mayor pet peeves with gqrx.
On my system (ArchLinux) the tooltips in the mainwindow showing the FFT and waterfall are black on black, which makes them almost impossible to read.

Apparently they get their color from setting `background-color: rgb(31, 29, 29);` in the Stylesheet of centralWidget  in mainwindow.ui

Creating a new Stylesheet for plotter with `QToolTip { background-color: #FFFFDD; }` fixes it, or at least makes the tooltips readable again. I have no clue how to tell QT to use its default tooltip style...

Alex

alexf91, 2017-04-22:
Arch is always fun to fix bugs ;-)
Which windowmanager are you using?

chron-isch, 2017-04-22:
i3.

alexf91, 2017-04-22:
Of course something strange^^
Maybe I'll find some time too look into it. I have essentially the same problem with Qtile.

csete, 2017-04-22:
On my computer it uses the system default (white text on black background)

chron-isch, 2017-04-22:
Like I said, I fixed it for myself by just adding `QToolTip { background-color: #FFFFDD; }` to the CPlotter stylesheet, but I really would like to figure out how to get it to just use the QT default settings.

LongnoseRob, 2017-04-22:
I had the same issue with unreadable tooltips on Arch w/ XFCE.

@chron-isch  you could try qt5ct (https://www.archlinux.org/packages/?name=qt5ct)
also helpful was reading https://wiki.archlinux.org/index.php/Qt#Configuration_of_Qt5_apps_under_environments_other_than_KDE_Plasma

csete, 2017-04-22:
FYI I set the background color of the centralWidget to rgb(31, 29, 29) - I wonder if that has any impact for you. It makes no difference for tooltips on my xfce.

chron-isch, 2017-04-22:
@csete That's excatly what I was thinking as well.

Screenshot: https://paste.xinu.at/WFW/

LongnoseRob, 2017-04-23:
@chron-isch after installing and running qt5ct (setting the colors you`d like), you need to set environment variable QT_QPA_PLATFORMTHEME=qt5ct, and then run gqrx.

chron-isch, 2017-04-23:
```
chron@censored # env | grep QT
QT_QPA_PLATFORMTHEME=qt5ct
```
Yep :)

LongnoseRob, 2017-04-23:
@chron-isch strange.. maybe i3wm is doing things different from the way xfce is doing

csete, 2017-04-23:
Which version of Qt are we talking about? There have been some changes from 5.6 to 5.7 regarding Qt themes. Qt 5.6 with xfce looks just fine; however, 5.7 and later looks like windows 95 unless I explicitly set the theme with "-style fusion" or something like that.

chron-isch, 2017-04-23:
qt5 5.8.0

csete, 2017-04-23:
So, I have built it with Qt 5.8 and have the same issue.

karstengit, 2017-06-09:
Hello gods of GQRX.
I have compiled GQRX V 2.6 for Debian stable and i have the same problem here.

Maybe the problem is caused in KDE by the used schema, but this problem occurs only in this application!
So how can i fix it?

I am not familiar with QT programming and i could not make the modifications as suggested, because they are not complete.
I tried it but i does not work!
@chron-isch : Please give a complete alternate file to compile!

This additional informations are a good new feature - but it is unusable now.
Please help to solve the problem.

remicles, 2017-06-22:
I had the same dark tooltips over waterfall and spectrum.  Linux Mint 17.1, Qt 5.2.1, GQRX v2.6-111-g82e0cc2.
This worked for me:
`export QT_STYLE_OVERRIDE=gtk`
Makes everything else look better too, IMO.

karstengit, 2017-06-23:
Thanks remicles.
This export really fix the problem.
```export QT_STYLE_OVERRIDE=gtk```

So there should be a fixed style for gqrx V 2.6

chron-isch, 2017-06-28:
@karstengit Sorry for the late reply, here's the patch im currently using:
```
--- src/gqrx-2.6.1/src/applications/gqrx/mainwindow.ui	2017-02-16 20:40:14.000000000 +0100
+++ src_mod/gqrx-2.6/src/applications/gqrx/mainwindow.ui	2017-04-23 00:49:58.702910208 +0200
@@ -153,6 +153,9 @@
         <verstretch>0</verstretch>
        </sizepolicy>
       </property>
+      <property name="styleSheet">
+       <string notr="true">QToolTip { background-color: #FFFFDD; }</string>
+      </property>
       <property name="frameShape">
        <enum>QFrame::StyledPanel</enum>
       </property>
```
But it's more of a workaround, since it just sets the background-color to something readable and not the default color every other tooltip uses.

stujones11, 2017-09-05:
Same here with debian 9, however, the workaround posted be @karstengit does fix the problem for me.

chyard, 2017-11-24:
Just wanted to thank everyone for this thread. Having the unreadable tooltips was a frustration. I tried all of the above env-based workarounds with no joy (Parrotsec current), but @karstengit's code patch solved the issue nicely. That's what open source is all about. Cheers.

dtwooton, 2018-02-23:
Here is a way to change the foreground

[tool_color.txt](https://github.com/csete/gqrx/files/1752642/tool_color.txt)

csete, 2018-03-16:
It's fixed and will be available in the next version / build!

tomswartz07, 2018-03-17:
Thanks @csete! I forgot how much I liked the tooltips, now that I can read them :)

closed 2018-03-18 by csete