gqrx-sdr/gqrx #513: Ubuntu PPA Builds

ddcc, 2017-04-23:
Is it possible to enable builds on the Ubuntu PPA for armhf? I was wondering why gqrx wasn't recognizing LimeSDR on an Odroid XU4, until I realized that there's no ARM build in the PPA and it was pulling an older version from the Ubuntu ports repository.

csete, 2017-04-23:
We had that before but it is a lot of trouble and I have no time. So unless you are volunteering to do the work it stays as it is.

Note that it also requires maintenance of gnuradio and other dependencies.

closed 2017-04-23 by csete

ddcc, 2017-04-23:
Ok, good to know. FWIW, at least on Ubuntu 16.04, gnuradio doesn't need to be compiled from source; 
using `gnuradio-dev` from the Ubuntu ports repository, and `libsoapysdr-dev`, `libsoapysdr-0.5-2`, and `soapysdr-module-lms` from the myriadrf/drivers PPA (armhf build), only `gr-osmocom` and `gqrx-sdr` need to be recompiled to use the soapysdr bindings.

csete, 2017-04-23:
We can not choose what to build for armhf and what not to build. We can only enable armhf for a PPA, then everything that is uploaded to that PPA is built for all supported platforms.

Anyway, I'd love to support this but somebody else will have to do it. I already spend way to much time on these kind of tasks instead of writing code.