gqrx-sdr/gqrx #518: connect() on localhost to rtl_sdr TCP service fails in an ECONNREFUSED loop

erikarn, 2017-05-07:
hi,

I'm running gqrx-2.6.1 on FreeBSD-HEAD with rtl-sdr-20140210 as the rtl-sdr userland. Both are FreeBSD packages.

rtl_tcp binds to 127.0.0.1 by default. However, gqrx tries to connect to 'localhost' which resolves first to the v6 address - so it tries to connect to '::1' - and it gets stuck in a connect() loop. It never tries the v4 address ;it doesn't give up after a few attempts and throw up an error and it doesn't log anything about the failed attempt.

csete, 2017-05-07:
What's your device string? Did you try setting the IP address?

Anyway, gqrx doesn't really do any interfacing to hardware. It uses the gr-osmosdr library so if there is a problem it's probably in gr-osmosdr. Note thatr rtl_tcp support is pretty much broken in the official gr-osmosdr library (see issue #99) so you will at least have to use our fork from https://github.com/csete/gr-osmosdr-gqrx (rtlsdr branch).

csete, 2017-05-07:
With device string I mean having something like

> rtl_tcp=127.0.0.1:1234

c.f. http://osmocom.org/projects/sdr/wiki/GrOsmoSDR#RTL-SDR-TCP-Source

erikarn, 2017-05-08:
hiya,

yup - if I specify 127.0.0.1, it works fine. It's just "localhost" that
doesn't and whatever's doing it gets stuck in a loop!

-adrian

csete, 2017-05-08:
Sounds good.

Closing this issue since there is nothing we can do in gqrx about this. The code which may be misbehaving is in the gr-osmosdr library.

closed 2017-05-08 by csete