gqrx-sdr/gqrx #519: Connect() loop silently failing with unreachable rtl_tcp address

martymac, 2017-05-09:
Hi,

When using a rtl_tcp device string with an IP address that is not reachable (or where no rtl_tcp server is listening), gqrx seems to fall down to an infinite connect() loop which is silently failing.

Tracing the process shows the following error :

 2537: connect(12,{ AF_INET6 [::1]:1234 },28)    ERR#61 'Connection refused'
 2537: connect(12,{ AF_INET6 [::1]:1234 },28)    ERR#61 'Connection refused'
 2537: connect(12,{ AF_INET6 [::1]:1234 },28)    ERR#61 'Connection refused'
 2537: connect(12,{ AF_INET6 [::1]:1234 },28)    ERR#61 'Connection refused'
[...]

I am using gqrx 2.6.1 on FreeBSD.

Thanks for your help,

Regards,

Ganael.

erikarn, 2017-05-09:
heh, I /just/ went through this with him! heh

martymac, 2017-05-09:
Great, thanks!

csete, 2017-05-09:
Don't know if you found the workaround, but it's in #518

closed 2017-05-09 by csete

csete, 2017-05-09:
Also, as I wrote in the other thread, the rtl_tcp interface in the official gr-osmosdr library is broken, so you will have to use my fork if you want to use rtl_tcp: https://github.com/csete/gqrx/issues/518#issuecomment-299742895

By the way, why do you want to use rtl_tcp on the same machine? Is there a problem with USB device access on FreeBSD?

martymac, 2017-05-09:
Thanks for the workaround in bug #518.

No, I have no problem with direct access to a local USB device, it works fine. I was just trying to reproduce bug #518 but was not aware of the fact that it had already been submitted (I got the bug through another channel).

Sorry for the noise... and thanks for developping Gqrx !

erikarn, 2017-05-09:
Oh, just wanting to get it working right first before i indeed use remote
rtl_tcp instances.

-a

csete, 2017-05-09:
Hey, no problem. I realize it can be difficult to know where a problem occurs since there are so many layers between gqrx and the hardware.