gqrx-sdr/gqrx #522: Frequency compensation only settable in integral ppm for original Funcube Dongle and Funcube Dongle Pro Plus

rag2, 2017-06-27:
The frequency correction set in GQRX to an FCD or FCDPP is only applied as integral ppm values, such as 1ppm, 2ppm, 3ppm &c. The consequence of this is that at 1296MHz, the compensation is quantised to ~1.30kHz. According to pull request 201, it **should** be possible to set the correction to 0.1ppm (the internal representation has much finer resolution). For weak signal narrow band working or monitoring, ~1.30kHz is an excessive quantisation step. The ability to set correction to 0.1ppm (~130Hz) or even 0.01ppm (~13Hz) is desirable operationally and would line up with the operation of other supported hardware.

**Presumed problem source**
The correction code in [https://github.com/gnuradio/gnuradio/blob/master/gr-fcd/include/gnuradio/fcd/source_c.h#L126](url) and [https://github.com/dl1ksv/gr-fcdproplus/blob/master/include/fcdproplus/fcdproplus.h#L90](url) is quantised to integral ppm in the calling interface. [Thanks to Alex for identifying this.]

**Possible solutions**

> Changing the API [in external libraries gr-fcd and gr-fcdproplus] might not be a good idea since it would break
binary compatibility. One could work around this in the gr-osmosdr
wrapper by applying the frequency correction here as floating point
number instead of calling the gr-fcd / gr-fcdpoplus functions.

> This is just an idea in case somebody wants to try. In the long run we
need a new "libfcd" driver library anyway.

> Alex

AsciiWolf, 2021-05-05:
I have the same issue on both Airspy R2 and RTL-SDR with Gqrx 2.14.4. Any chance this could be fixed?

AsciiWolf, 2021-05-05:
Sorry for the noise, I was testing this incorrectly, it seems to work fine on Airspy R2. It only seems to have problems with RTL-SDR.

yarda, 2022-02-06:
I am afraid there is nothing to fix on the gqrx or gr-osmosdr side, both use double. The driver (currently the gr-funcube, because both gr-fcdproplus and gr-fcd are obsoleted) needs to be fixed. I opened gr-funcube bug https://github.com/dl1ksv/gr-funcube/issues/5.

yarda, 2022-02-06:
Regarding the rtlsdr, if I am not mistaken, it seems the HW could do 1/16 steps, which is definitely better than the 1 ppm steps.

yarda, 2022-02-06:
For the rtlsdr the gr-osmosdr uses int in the wrapper, i.e. for the rtlsdr the gr-osmosdr also needs patching. Nevertheless, I opened the rtlsdr bug https://github.com/steve-m/librtlsdr/issues/66 because I think the driver should use the full capabilities of the HW.

Regarding the airspy I didn't check the code, nor I have/know the device.

Maybe the frequency could be corrected solely on the gqrx side and the drivers set_freq_corr() needn't be used.

yarda, 2022-02-17:
This has been fixed in the gr-funcube and I also filled gr-osmosdr issue with the patch. We patched it in the Fedora downstream and for the FUNcube Dongle and FUNcube Dongle Pro+ it will start working in the Fedora out-of-the-box, i.e. no changes needed on the gqrx side for it.