gqrx-sdr/gqrx #523: Create new Tag in bookmarks doesn't work

georgievgeorgi, 2017-06-27:
Steps to reproduce the problem:
- Right click on Create new Tag in tags box
- Give it a name and confirm
- Click on some other tag
In this moment when the list gets refreshed the newly made tag disappears.

csete, 2018-03-17:
Yes, I see this. It is still possible to create a new tag when adding a new bookmark.

kolbma, 2020-03-11:
@georgievgeorgi 

I think this would be fixed in the changeset https://github.com/kolbma/gqrx/tree/bookmarks-ext
Although I've seen that I need one more change, that it doesn't edit the last tag in list any longer, but the newly added. Tomorrow ;-)

darrenfreeman, 2020-10-14:
For a long time now, gqrx has segfaulted for me, after creating a new tag. I can only reliably create them by editing the bookmarks.csv manually.