gqrx-sdr/gqrx #524: Feature Request: Remote Control LNA Gain

danhosler, 2017-06-27:
I would like to be able to control my dongle's gain via remote control

csete, 2017-08-19:
We can take a look, but I 'm not sure how to handle this. Different devices have different type of gain settings.

danhosler, 2017-08-19:
only toggling the gain on/off might be easier if you need to accommodate all devices.

alexf91, 2017-08-19:
I've started working on this.
I'll expose it through the "level" command (l, L) and call the different gain settings <GAIN_NAME>_GAIN.

csete, 2017-08-19:
What I was trying to say is that different devices have different gain names, so the TCP client would need to know which gains are available. Then suddenly we have data flow from gr-osmosdr to a TCP client - sounds like a bad idea...

alexf91, 2017-08-19:
I haven't really thought about the client side, but the available gain names can be queried by using the "l ?" command and filtering the return values for "_GAIN".

csete, 2020-01-26:
PR #635 merged and will be available in 2.12.

closed 2020-01-26 by csete