gqrx-sdr/gqrx #525: RTL-SDR support broken

aquila, 2017-06-30:
Hi,

I played successfully with an RTL-SDR on OSX a couple of days ago, allowing me to listen to FM broadcast radio. After failing to move my RTL-SDR to a Pi with a connection over TCP, I decided to return to my laptop. Upon removing the remote IP, local RTL-SDR does not seem to work anymore. Even just starting gqrx sometimes hangs, unless I pick the IQ file option. Then, selecting my stick also hangs the program. I tried rebooting my laptop, which did not solve the issue. Also removing gqrx and installing it again from the DMG fails.
I also tried removing everything in .config/gqrx and then reinstalling the program, which failed too.
Did I somehow break my settings?

aquila, 2017-06-30:
From the hang log:

Date/Time:       2017-06-30 08:34:04 +0200
OS Version:      Mac OS X 10.12.5 (Build 16F73)
Architecture:    x86_64h
Report Version:  25

Data Source:     Stackshots

Command:         gqrx
Path:            /Applications/Gqrx.app/Contents/MacOS/gqrx
Version:         2.6 (???)
Parent:          launchd [1]
PID:             1635

Event:           hang
Duration:        1.10s (process was unresponsive for 15 seconds before sampling)
Steps:           11 (100ms sampling interval)

Hardware model:  MacBookPro12,1
Active cpus:     4

Fan speed:       0 rpm

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  11  start + 1 (libdyld.dylib + 21045) [0x7fffbf9a9235]
  11  main + 5147 (gqrx + 29803) [0x109fb846b]
  11  MainWindow::MainWindow(QString, bool, QWidget*) + 6949 (gqrx + 56997) [0x109fbeea5]
  11  CIoConfig::getDeviceList(std::__1::map<QString, QVariant, std::__1::less<QString>, std::__1::allocator<std::__1::pair<QString const, QVariant> > >&) + 2655 (gqrx + 1200079) [0x10a0d5fcf]
  11  ??? (<FF9237CE-C285-3600-A838-4380ECD79190> + 174151) [0x10ad6b847]
  11  ??? (<FF9237CE-C285-3600-A838-4380ECD79190> + 317181) [0x10ad8e6fd]
  11  ??? (<75889FFF-206A-3495-AAE8-F13EEF2F6E29> + 7893) [0x10c3c3ed5]
  11  ??? (<B7983B3E-FDC7-3480-B982-2BDB933648A2> + 10132) [0x10c40f794]
  11  __semwait_signal + 10 (libsystem_kernel.dylib + 106310) [0x7fffbfad7f46]
 *11  semaphore_wait_continue + 0 (kernel + 1111056) [0xffffff800030f410]

Process:         gqrx [1635]
Path:            /Applications/Gqrx.app/Contents/MacOS/gqrx
Architecture:    x86_64
Parent:          launchd [1]
UID:             501
Task size:       59.16 MB (-64 KB)
CPU Time:        <0.001
Note:            Unresponsive for 15 seconds before sampling
Note:            3 idle work queue threads omitted

  Thread 0x10278            DispatchQueue 1           11 samples (1-11)         priority 46 (base 46)
  <thread QoS user interactive (requested user interactive), process unclamped, IO tier 0>
  11  start + 1 (libdyld.dylib + 21045) [0x7fffbf9a9235] 1-11
    11  main + 5147 (gqrx + 29803) [0x109fb846b] 1-11
      11  MainWindow::MainWindow(QString, bool, QWidget*) + 6949 (gqrx + 56997) [0x109fbeea5] 1-11
        11  CIoConfig::getDeviceList(std::__1::map<QString, QVariant, std::__1::less<QString>, std::__1::allocator<std::__1::pair<QString const, QVariant> > >&) + 2655 (gqrx + 1200079) [0x10a0d5fcf] 1-11
          11  ??? (<FF9237CE-C285-3600-A838-4380ECD79190> + 174151) [0x10ad6b847] 1-11
            11  ??? (<FF9237CE-C285-3600-A838-4380ECD79190> + 317181) [0x10ad8e6fd] 1-11
              11  ??? (<75889FFF-206A-3495-AAE8-F13EEF2F6E29> + 7893) [0x10c3c3ed5] 1-11
                11  ??? (<B7983B3E-FDC7-3480-B982-2BDB933648A2> + 10132) [0x10c40f794] 1-11
                  11  __semwait_signal + 10 (libsystem_kernel.dylib + 106310) [0x7fffbfad7f46] 1-11
                   *11  semaphore_wait_continue + 0 (kernel + 1111056) [0xffffff800030f410] 1-11

  Thread 0x102a4            Thread name "org.libusb.device-hotplug"             11 samples (1-11)         priority 31 (base 31)
  <thread QoS default (requested default), process unclamped, IO tier 0>
  11  thread_start + 13 (libsystem_pthread.dylib + 12429) [0x7fffbfbc208d] 1-11
    11  _pthread_start + 286 (libsystem_pthread.dylib + 14471) [0x7fffbfbc2887] 1-11
      11  _pthread_body + 180 (libsystem_pthread.dylib + 14651) [0x7fffbfbc293b] 1-11
        11  ??? (<B7983B3E-FDC7-3480-B982-2BDB933648A2> + 54880) [0x10c41a660] 1-11
          11  CFRunLoopRun + 97 (CoreFoundation + 813969) [0x7fffaa26eb91] 1-11
            11  CFRunLoopRunSpecific + 420 (CoreFoundation + 553236) [0x7fffaa22f114] 1-11
              11  __CFRunLoopRun + 1361 (CoreFoundation + 555201) [0x7fffaa22f8c1] 1-11
                11  __CFRunLoopServiceMachPort + 212 (CoreFoundation + 558132) [0x7fffaa230434] 1-11
                  11  mach_msg_trap + 10 (libsystem_kernel.dylib + 74570) [0x7fffbfad034a] 1-11
                   *11  ipc_mqueue_receive_continue + 0 (kernel + 854224) [0xffffff80002d08d0] 1-11

  Binary Images:
           0x109fb1000 -        0x10a167fff  dk.gqrx.www 2.6 (0)                   <33F1B86A-DE06-3936-A5DA-4C66BBAF8926>  /Applications/Gqrx.app/Contents/MacOS/gqrx
           0x10ad41000 -                ???  ???                                   <FF9237CE-C285-3600-A838-4380ECD79190>
           0x10c3c2000 -                ???  ???                                   <75889FFF-206A-3495-AAE8-F13EEF2F6E29>
           0x10c40d000 -                ???  ???                                   <B7983B3E-FDC7-3480-B982-2BDB933648A2>
        0x7fffaa1a8000 -     0x7fffaa641fff  com.apple.CoreFoundation 6.9 (1349.8) <09ED473E-5DE8-307F-B55C-16F6419236D5>  /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
        0x7fffbf9a4000 -     0x7fffbf9a9fff  libdyld.dylib (433.5)                 <EC3D88D2-3D40-3274-8E26-362C2D7352C8>  /usr/lib/system/libdyld.dylib
        0x7fffbfabe000 -     0x7fffbfae0fff  libsystem_kernel.dylib (3789.60.24)   <6E9E485F-91F6-36B7-A125-AE91DC978BCC>  /usr/lib/system/libsystem_kernel.dylib
        0x7fffbfbbf000 -     0x7fffbfbc9fff  libsystem_pthread.dylib (218.60.3)    <B8FB5E20-3295-39E2-B5EB-B464D1D4B104>  /usr/lib/system/libsystem_pthread.dylib
   *0xffffff8000200000 - 0xffffff80009fffff  kernel (3789.60.24)                   <FB2170A8-257D-3C64-B14D-BC06CC94E34C>  /System/Library/Kernels/kernel

csete, 2017-06-30:
Have you seen the existing bug reports and discussions on the mailing list regarding running gqrx on Mac OS X 10.12?

The only thing I can suggest is to run gqrx  and rtl_test (included in the bundle) from the command line to see if they print any useful error message.

Sorry I can't be of more help, but as I have written on other discussions I can neither build nor test gqrx on Mac OS X 10.12.

dsward2, 2017-07-22:
A couple of ideas to try if Mac has problems launching gqrx -

1. Use Terminal.app to cd into the gqrx.app bundle, then run the MacOS/gqrx executable from the command line.  If that works, it might be helpful in repairing the settings, and perhaps it will launch normally the next time.

2. If using the installable version of gqrx and problems continue, try the MacPorts version of gqrx that includes the latest source code updates.

vk4tec, 2017-07-22:
Mine always blocks gqrx says it's an unknown author

aquila, 2017-07-24:
Terminal hangs too, strange...

aquila, 2017-07-24:
Commandline (via Homebrew) hangs too.