gqrx-sdr/gqrx #527: Plugins and SNR

ruy-benton, 2017-07-05:
Thank you for your wonderful software.

Several app extend functionality with Plugins and Extensions.
It's possible?

I would love to see SNR in main screen.

Thak you
Ruy

csete, 2017-07-06:
Hi Ruy,

Plugins are not supported by gqrx but SNR reading might be added later.

alexf91, 2017-07-06:
I actually tried to add functionality for Python plugins about a year ago. It didn't really go anywhere though. It was just something I was playing around with for a weekend or so.
As far as I remember, I started with PythonQt but switched to a more low-level Python implementation, although I can't remember why.

The idea behind it was to provide an interface for data decoders with a user interface integrated in Gqrx. I thought the idea was quite interesting, but I got distracted by other projects. Maybe I can find the branch on a backup tomorrow.

csete, 2017-07-06:
@alexf91 

I think adding plugin support in the sense of loadable modules is a bad idea. The gqrx architecture is already very complex and fragile with things breaking all the time.

Better support and integration of data decoders could be achieved by launching console based applications from within gqrx and interacting with them via stdin/stdout/stderr. The mechanism for doing this is already available in Qt.

alexf91, 2017-07-06:
@csete I thought so too after I read about that. For completeness, #100 is a feature request that addresses this.

martinber, 2018-11-02:
Just in case someone finds this googling, it's possible to [stream audio using UDP into another program](http://gqrx.dk/doc/streaming-audio-over-udp). Not exactly a plugin but close enough, on that link you can already see examples of programs taking advantage of this.