gqrx-sdr/gqrx #529: Cannot update 2.5.3 -> 2.6.1 in NixOS/nixpkgs

bjornfor, 2017-07-15:
I'm a gqrx maintainer in NixOS/nixpkgs and we are having trouble upgrading to 2.6.1. What happens is that in version 2.6.1 the buttons in the menu bar are all gray. The buttons can be clicked, but they are gray and have no icon. 

You can see the discussion at https://github.com/NixOS/nixpkgs/pull/24577 and my latest update attempt at https://github.com/bjornfor/nixpkgs/tree/gqrx2.

I realize this may be entirely our own problem, but I thought I'd mention it here anyway.

csete, 2017-07-15:
Hi @bjornfor 

You need to have the QtSvg package installed, otherwise gqrx can't load the SVG icons. So please add that library as a runtime dependency.

bjornfor, 2017-07-15:
Hi @csete. You were right. QtSvg was not picked up at runtime. Explicitly installing it fixed the problem.

I'll close this and continue the discussion in NixOS/nixpkgs, because generally build time dependencies become available at runtime too, but obviously this didn't work for the QtSvg case.

csete, 2017-07-15:
This is a problem on all platform. QtSvg is a plugin that is not linked to at build time, so it must be explicitly added as a dependency.

bjornfor, 2017-07-15:
I didn't know that. Thanks.

bjornfor, 2017-07-16:
It's weird though that none of the other packages in nixpkgs that depend on qtsvg seem to be doing anything special to have it available at runtime. Could it be that _gqrx_ requires this explicit runtime handling on all platforms, whereas other programs (like "subsurface") somehow manages to retain this dependency from the build environment automatically?

Looking closer at "subsurface", a qt5 application using Qt5Svg and built with cmake.

```
$ readelf -a /nix/store/bznqn77z5y9z1hwz6invi89gmkz5b34f-subsurface-4.6.0/bin/subsurface | grep NEEDED | grep libQt
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Concurrent.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Svg.so.5]
 ...
```

So it does have explicit libQt5Svg dependency in its ELF file (whereas gqrx does not).

Looking at subsurface's CMakeLists.txt file, I see this

```
find_package(Qt5 REQUIRED COMPONENTS Core Concurrent Widgets Network Svg Test LinguistTools Positioning ${QT_EXTRA_COMPONENTS})
set(QT_LIBRARIES Qt5::Core Qt5::Concurrent Qt5::Widgets Qt5::Network Qt5::Svg  Qt5::Positioning ${QT_EXTRA_LIBRARIES})
```

followed by

```
target_link_libraries(<program_here> ${QT_LIBRARIES})
```

So that's how they solve it. Perhaps gqrx can do the same?

(Background: NixOS/nixpkgs users like myself dislike having to install non-leaf programs/libraries just to satisfy dependencies, if these dependencies could otherwise be automatically handled in a side by side, non-interfering manner. Installing qtsvg "globally" for gqrx means that is the only qtsvg version that can exist (globally) on the system. If another application with a similar need comes along, but requiring different qt version, we have a problem.)

csete, 2017-07-16:
I tried it but it doesn't seem to make any difference. Note that the other applications might be using QtSvg directly whereas for gqrx it is an indirect dependency.

I will try to add a dummy function call to QtSvg to enforce the link. In the meanwhile, feel free to try your method and see if it is sufficient.

csete, 2017-07-16:
It looks like it works if I create a dummy SVG object. I pushed a commit with this change.

If you want, I can try to cherry-pick it into the 2.6 branch, or perhaps you can apply the patch yourself while packaging?

bjornfor, 2017-07-17:
For me, the dummy object made no difference. To get the NEEDED entry in the ELF, only target_link_libraries(prog ${QT_LIBRARIES}) was needed. But that is apparently not enough to make it work.

So what changed on your side when you added the dummy object? Didn't you get the libQtsvg NEEDED entry in the ELF without it?

Do you know how to debug the qtsvg plugin mechanism? Why isn't qtsvg working by simply having libQtsvg library linked to the ELF?

Now that I think about it, I cannot be _sure_ that "subsurface" works (my reference application). I know it links to libQtsvg, but I'm not familiar with its UI and perhaps I simply haven't seen any of the UI elements that qtsvg is used for.

I'll ask Qt maintainer in NixOS/nixpkgs.

I can apply patches myself for packaging, when this is resolved.

bjornfor, 2017-07-17:
I've looked into how the Qt plugin mechanism is implemented in nixpkgs and have found a way to make qtsvg available to gqrx without "global" install. (I'm not sure the end result provides any better _isolation_, but it still feels slightly better this way.)

In the end it was a nixpkgs issue. Although depending on qtsvg at build time doesn't hurt :-)

So, case closed. Thanks!

closed 2017-07-17 by bjornfor

csete, 2017-07-19:
Thanks for the update!

paulrames, 2017-09-03:
Hi!

After upgrading to version 2.7, I noticed I could no longer click on the "Start DSP" or "Configure I/O" toolbar buttons. Only the very bottom of the buttons is clickable. This affects, at least, Ubuntu 14.04. I tried different computers (all with Ubuntu) to make sure this was not some configuration problem.

Git-bisect led me to 75c8ee79, related to this issue. It seems the dummy widget is placed over the first two toolbar buttons, and it doesn't get removed from the screen. Displaying an SVG image inside it helps to visualize the problem:

[image]

I found that creating the widget without a parent solves the problem for me, but I don't know if it's the best way:

`qsvg_dummy = new QSvgWidget();`

instead of

`qsvg_dummy = new QSvgWidget(this);`

Thanks a lot

csete, 2017-09-03:
Heh,I though I fixed this, but apparently not. Thanks for the heads up @paulrames - now fixed in git.