gqrx-sdr/gqrx #530: compiling with cmake gets an AppenderMapStorageInitializer error

hkopp64, 2017-07-17:
Hello,

After almost ending with compiling in cmake, I get undelaying error message and searching about this on google but gets me not further te solve the problem.
Does someone earlier dealing with this issue and know about to solve it.
Many thanks in advance.

Huib

[ 89%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/receiver_base.cpp.o
[ 90%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/wfmrx.cpp.o
[ 92%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_device_list.cc.o
/usr/local/src/gqrx/src/pulseaudio/pa_device_list.cc: In member function ‘int pa_device_list::populate_device_list()’:
/usr/local/src/gqrx/src/pulseaudio/pa_device_list.cc:129:49: warning: ‘pa_op’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                 if (pa_operation_get_state(pa_op) == PA_OPERATION_DONE) {
                                                 ^
[ 93%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_sink.cc.o
[ 94%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_source.cc.o
[ 96%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_icons.cpp.o
[ 97%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_textfiles.cpp.o
[ 98%] Building CXX object src/CMakeFiles/gqrx.dir/gqrx_automoc.cpp.o
[100%] Linking CXX executable gqrx
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/gqrx.dir/applications/gqrx/main.cpp.o: undefined reference to symbol '_ZN7log4cpp8Appender29AppenderMapStorageInitializerD1Ev'
/usr/lib64/liblog4cpp.so.5: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/CMakeFiles/gqrx.dir/build.make:1634: recipe for target 'src/gqrx' failed
make[2]: *** [src/gqrx] Error 1
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/gqrx.dir/all' failed
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

csete, 2017-07-17:
The error seems to come from liblog4cpp.so.5, which is not a direct dependency of gqrx. AFAIK gqrx is already available on suse, so perhaps you can see if they had to patch something for compiling. Otherwise please see http://gqrx.dk/user-support for support. As written there, we use github issues only for bug tracking only and not as a support forums. Thanks.

closed 2017-07-17 by csete

vk4tec, 2017-07-17:
Having said that, what is the preferred LINUX flavour for GQRX ?

 

Andrew

csete, 2017-07-17:
There is no preferred Linux flavour for Gqrx. We provide binary packages for Ubuntu Linux, which can also be used by other flavors based on Ubuntu, but most distribution already have binary packages either through the official channels or through 3rd party repositories.

hkopp64, 2017-07-18:
csete,

The depended files in suse are liblog4cpp5 and log4cpp-devel and are available (if avaailable) in the library /usr/lib64/
But on these files there is no testing taken place in the cmake file.
Where do i have to put liblog4cpp.so.5 in the cmake filelist (I gues source dir of gqrx) so that it will be find for the make file proces.
Thanks in advance

Huib

csete, 2017-07-18:
@hkopp64 

As I have said, gqrx neither use nor depend on liblog4cpp, so the error is in the configuration of one of the gnuradio libraries.

mnhauke, 2017-07-18:
@hkopp64 

There are some issues with log4cpp and GNU Radio.
Please have a look at the following GNU Radio issue I opened last year: https://github.com/gnuradio/gnuradio/issues/1045
A proper fix will make it into GNU Radio 3.8

You probably do not need GNU Radio with log4cpp enabled and you seems to use openSUSE I see the following options:
* in case you installed GNU Radio from sources yourself just do a rebuild without having lib4cpp enabled (uninstall log4cpp-devel first)
* use gqrx,gnuradio and all the other sdr-related stuff from the hardware:sdr repository:
https://build.opensuse.org/project/show/hardware:sdr