gqrx-sdr/gqrx #532: [Feature Request] disable waterfall pane

jurgynet, 2017-07-27:
Would be nice to have waterfall in the view/not view options.
for when running over network disabling all that drawing would be nice, specially when just using for FM at times.

csete, 2017-07-27:
I'm marking it as a feature, but don't expect to see this implemented anytime soon. The whole purpose with gqrx was to provide a nice graphical SDR application.

IMO, simply hiding the FFT would make the GUI look awkward, since there would be a big empty rectangle, unless we also rework how the control panels are placed. Then suddenly we are talking about more work than a simple show/hide functionality.

Also note that you can disable both FFT processing and drawing by setting the FFT rate to 0.

vk4tec, 2017-07-27:
Have you guys ever noticed you can scroll the spectrum display and go into black space ?

jurgynet, 2017-07-27:
Thanks csete, Setting FFT to 0fps escaped my notice and accomplishes my need. I'm running X11 apps over my network since the Hifi is across the room... Jurgyman.

csete, 2017-07-27:
@jurgynet Ok, glad to hear that. Stay tuned because soon I will release a new SDR that can be used over network.

csete, 2017-07-27:
@vk4tec 

> Have you guys ever noticed you can scroll the spectrum display and go into black space ?

Yes, I am aware of this, but that's a different topic, so lets not hijack the thread.

LightningStalker, 2019-09-19:
Monitoring a single frequency in the background causes enough CPU load to increase fan speed. Setting rate to 0 fps reduces this by only about half.