gqrx-sdr/gqrx #537: Latest release crashes at startup on Mac OS 10.11.6

hemna, 2017-08-18:
I just downloaded the latest from the main website and tried to fire it up on my mac.

[gqrx-log.txt](https://github.com/csete/gqrx/files/1235693/gqrx-log.txt)

csete, 2017-08-18:
What kind of mac?
There is a known issue with Mac Pros and even some workaround, see issue #324

vk4tec, 2017-08-22:
MacBook:src vk4tec$ sudo port uninstall gqrx
Password:
--->  Deactivating gqrx @2.6.1_2
--->  Cleaning gqrx
--->  Uninstalling gqrx @2.6.1_2
--->  Cleaning gqrx

sudo git clone < source.git >
cd gqrx
sudo mkdir build 
cd build 
qmake ..

< now i have no qmake on my mac >

Looking at why this is 

Will report back - Mac OS X latest - Mac Book Pro

Andrew

vk4tec, 2017-08-22:
MacBook:bin vk4tec$ sudo port install gqrx
--->  Computing dependencies for gqrx
--->  Fetching archive for gqrx
--->  Attempting to fetch gqrx-2.6.1_2.darwin_16.x86_64.tbz2 from http://nou.nc.packages.macports.org/pub/macports/packages.macports.org/gqrx
--->  Attempting to fetch gqrx-2.6.1_2.darwin_16.x86_64.tbz2 from http://kmq.jp.packages.macports.org/gqrx
--->  Attempting to fetch gqrx-2.6.1_2.darwin_16.x86_64.tbz2 from http://jog.id.packages.macports.org/macports/packages/gqrx
--->  Fetching distfiles for gqrx
--->  Verifying checksums for gqrx
--->  Extracting gqrx
--->  Applying patches to gqrx
--->  Configuring gqrx
--->  Building gqrx
--->  Staging gqrx into destroot
--->  Installing gqrx @2.6.1_2
--->  Activating gqrx @2.6.1_2
--->  Cleaning gqrx
--->  Updating database of binaries
--->  Scanning binaries for linking errors
--->  No broken files found.                             
MacBook:bin vk4tec$ gqrx
Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_105900; UHD_003.010.002.000-MacPorts-Release

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy soapy 
gr::fft: can't import wisdom from /Users/vk4tec/.gr_fftw_wisdom
gr::log :INFO: audio source - Audio sink arch: osx
2017-08-23 03:43:33.998 gqrx[2320:71461] Error loading /Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter:  dlopen(/Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter, 262): Library not loaded: /System/Library/Frameworks/Jackmp.framework/Versions/A/Jackmp
  Referenced from: /Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter
  Reason: image not found
2017-08-23 03:43:33.998 gqrx[2320:71461] Cannot find function pointer New_JackRouterPlugIn for factory 7CB18864-927D-48B5-904C-CCFBCFBC7ADD in CFBundle/CFPlugIn 0x7fb695637ce0 </Library/Audio/Plug-Ins/HAL/JackRouter.plugin> (bundle, not loaded)
gr::log :INFO: audio_osx_sink0 - 

Using output audio device 'Built-in Output'.
  ... which is the current default output audio device.
  Changing the default output audio device in the System Preferences will 
  result in changing it here, too (with an internal reconfiguration).

BookmarksFile is /Users/vk4tec/.config/gqrx/bookmarks.csv
Number of audio input devices: 4
Number of audio output devices: 2
**Segmentation fault: 11**
MacBook:bin vk4tec$

vk4tec, 2017-08-22:
Ok only this one works on Mac 

https://github.com/csete/gqrx/releases/download/v2.6/Gqrx-2.6-1.dmg

vk4tec, 2017-08-22:
Source code build causes segmentation fault 

Andrew

csete, 2017-08-22:
Give me break...

closed 2017-08-22 by csete

hemna, 2017-08-24:
Mac Pro (early 2009)  
OS X El Capitan version 10.11.6

csete, 2017-08-24:
Ok, sorry I closed this but the issue suddenly got hijacked. Anyway, the problem you have seems to be the same as in issue #324.