gqrx-sdr/gqrx #540: AFSK 1200 decode - add TCP tocket to APRS-IS

vk4tec, 2017-08-19:
gqrx remains THE best program for LINUX and Mac OS X

The 1200 baud AFSK decoder is perfect and the doppler freq shift lends itself to a perfect satellite ground station.

The only thing missing now is the forwarding of the packets to the APRS-IS to contribute packets to other stations.

I know there is direworlf but I would like to increase the value of gqrx.

I dont think this would take much code to add a TCP socket and would establish gqrx as the only SDR software that is most popular and usable for Raspberry Pi

Getting more people interested in I Gates and Sat Gates

I am willing to have a go at the code 

Andrew VK4TEC

csete, 2017-08-19:
Hi Andrew,

My philosophy has always been to keep things simple and down to basics, so interfacing to igates and such is outside the scope of gqrx. In fact, the AFSK1200 decoder was never intended to be anything else than an experiment and should never have been included in gqrx. It's not even a very good afsk decoder from a performance point of view.

I fully understand the convenience of having data decoders integrated and launched from within the application, but I think there are better ways to achieve that than implementing everything inside gqrx. Issue #100 covers that.

You are of course welcome to modify gqrx to include such functionality. That's why it is open source.

closed 2017-08-19 by csete

TheButterZone, 2019-02-12:
OSX 10.13.6/Gqrx 2.11.5 user here, hoping to see an easy-peasy IGate for OSX to be written. By comparison, getting Direwolf set up on OSX is a nightmare for non-Mensa members. I had been using AFSK1200 Decoder.app with CubicSDR, a Behringer mixer & my Nooelec NESDR SMArt before even downloading Gqrx.

I have a Pi but it's tied up as a Motion server & I need one NESDR for tuning on my Mac anyway, so I'd have to buy another set to make an IGate.