gqrx-sdr/gqrx #541: Problems using gqrx with linux debian 9.1 and adalm-pluto using branch master

G4DCP, 2017-08-19:
I get a strange error when I try to use a USB connected adalm-pluto:-

Unable to write attribute out_altvoltage0_RX_LO_frequency: -22

I'm not sure if the problem is in gr-osmosdr or gqrx.

I have used the gr-osmosdr plutsdr branch and this works fine with test calls from grc.

Is this a known issue? Any recommendations that I can follow to try and get this working?

Thanks in advance for your help 
- Peter (G4DCP)

PS I am very impressed by gqrx

csete, 2017-08-19:
I also get that message but it still works. I'm confused by why it works in grc but not in gqrx. Are you sure you are using the gr-osmosdr source in grc?

G4DCP, 2017-08-21:
Yes I have built it using the gr-osmosdr sources I think my problem is setting the hardware base frequency rather than anything else I will have another look and get some more diagnostics. 

 

Many thanks for confirming that that message is not the reason I am having problems.

G4DCP, 2017-08-21:
OK,

As usual it was an operator issue, the message is still present but after playing with settings I now have it working.

Many thanks now to play awhile.

closed 2017-08-21 by G4DCP