gqrx-sdr/gqrx #542: sndio support

reezer, 2017-08-20:
Currently gqrx only supports Pulseaudio and Portaudio. It would be nice, if [sndio](http://www.sndio.org/) would also be supported.

csete, 2017-08-20:
Exactly why do we need it?

Sorry, but it has always been against my philosophy to add a bunch of different implementations that may or may not be needed. As far as I know, gqrx works just fine on BSD systems, no?

I may at some point switch to use the Qt Audio API and let that handle all the native interfaces.

reezer, 2017-08-20:
I am using sndio on various systems (both Linux and FreeBSD).

Using the Qt Audio API sounds like a great solution though.

csete, 2017-08-20:
Ok, we will see.

gvanem, 2017-12-18:
I'd vote for [BASS](http://www.un4seen.com/) instead. Very feature rich. Besides, [cuSDR](https://github.com/hvh/cusdr3) uses it.