gqrx-sdr/gqrx #543: gqrx fails to start with QXcbConnection error

lr101fc, 2017-08-22:
Fresh install of Raspbian stretch on a clean micro sd card on a Pi3
USB keyboard, mouse, ethernet network cable, audio to powered speakers and HDMI to a dell display
power from external power supply

Followed the install instructions on the webpage

sudo apt-get update
$ sudo apt-get upgrade
   (reboot if necessary)
$ sudo apt-get install libqt5gui5 libqt5core5a libqt5network5
$ sudo apt-get install libqt5widgets5 libqt5svg5 libportaudio2
$ wget https://github.com/csete/gqrx/releases/download/v2.6/gqrx-2.6-rpi3-2.tar.xz
$ tar xvf gqrx-2.6-rpi3-2.tar.xz
$ cd gqrx-2.6-rpi3-2
$ ./setup_gqrx.sh

logged in as user pi

pi@raspberrypi:~/gqrx-2.6-rpi3-2 $ sh run_gqrx.sh
run_gqrx.sh: 4: [: Illegal number:
Starting gqrx...
QXcbConnection: Could not connect to display
Aborted

Is this a raspbian pi3 with stretch issue or am I missing something else ?

thanks

Robin

vk4tec, 2017-08-22:
Robin are you running your raspberry pi in Desktop ?

It seems to want to connect to a display that should be running ?

Are you just installing this from command line only ?

Andrew

lr101fc, 2017-08-22:
Hi Andrew

thank you for your suggestion, yes I was running in cli only mode
now started the startx desktop and issued the sh run_gqrx.sh command from a prompt within the window and gqrx has started

thanks again for your help !
Robin

closed 2017-08-22 by lr101fc