gqrx-sdr/gqrx #547: Feature Request: Sent LNB LO via remote control

sdonnan, 2017-08-26:
I'm currently using the GQRX software with an RTL-SDR dongle as a pandapter for my radio. This setup works quite well. I set the LNB LO frequency in the configuration dialog for the first I/F of my radio, and then once I am running I can use the LNB LO display frequency adjustment in the Input Controls panel to compensate the displayed frequency on the UI.

The issue is that when I change the tuning of my radio I must manually recalculate and enter the display LNB LO frequency for the new config. This would be straightforward to automate by polling over the CAT interface to the radio and then setting the display LNB LO in GQRX via the remote interface. I am requesting the addition of a remote command to implement this functionality.

Note that I do not want to change the LNB LO used for tuning the RTL-SDR, I just want to change the LNB LO used for calculating the displayed frequency. Not sure what the best terminology is to distinguish between the two.

I am willing to take a stab at implementing this if you approve of the idea. I'd rely on you and others with experience interfacing other software to select the syntax of the command so I don't inadvertently create a compatibility problem.

alexf91, 2017-08-26:
I thought this was already possible, because I've been also using Gqrx to monitor my radio's IF during a contest once.

It looks like I've implemented this more than a year ago, but forgot to make a pull request.
It's not possible to merge now because of some refactoring in the remote control code.
I'm also not sure if it is buggy or not.

Here is my implementation:
https://github.com/alexf91/gqrx/tree/remote-lnb-settings

The signal connections should be fine, but the code in remote_control.cpp needs to be updated.
If you want to continue working on this, you could at least use it as a starting point.

Let's wait what @csete has to say about this.

csete, 2017-08-26:
Sounds like it could be useful, so just go ahead and do it, regardless of what I think ;-)

All I ask is that you check the hamlib commands to see whether there already is a command that could be used instead of introducing a new one.

I need to focus on some other projects for the rest of the year and I will probably be a bit slow with my responses.

sdonnan, 2017-08-26:
Sounds good. I'll start taking a look at the source. Might take me a couple weekends before I have anything to show for it.

@alexf91 Thanks for blazing the trail. Sounds like your branch might be a good starting point.

closed 2017-08-27 by csete