gqrx-sdr/gqrx #555: undefined symbol: rtlsdr_set_bias_tee

ikizhvatov, 2017-09-04:
Doing install in Ubuntu exactly as described in http://gqrx.dk/download/install-ubuntu leads to a symbol lookup error upon startup with RTL-SDR. Ubuntu version according to `lsb_release -a` is 16.04.3 LTS.

```
~$ gqrx
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-release

Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd plutosdr miri hackrf bladerf rfspace airspy soapy redpitaya 
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/ilya/.config/gqrx/bookmarks.csv
getDeviceList : Available input devices:
    0 : "Realtek RTL2838UHIDIR SN: 00000001"
    1 : "PlutoSDR"
    2 : "RFSPACE SDR-IQ Receiver"
    3 : "RFSPACE SDR-IP Receiver"
    4 : "RFSPACE NetSDR Receiver"
    5 : "RFSPACE Cloud-IQ Receiver"
    6 : "RTL-SDR Spectrum Server"
    7 : "Red Pitaya Transceiver Server"
    8 : "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/ilya/.config/gqrx/default.conf"
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd plutosdr miri hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
gqrx: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.1.5git: undefined symbol: rtlsdr_set_bias_tee
```

csete, 2017-09-04:
You have to remove libraries that you have installed from source before using the PPA. If you need further help, please don't use the bug tracker but follow the instructions here: http://gqrx.dk/user-support

closed 2017-09-04 by csete

chb0github, 2017-09-27:
@csete did you get this resolved? I have an identical issue and I installed fresh (so, no previous drivers)

ikizhvatov, 2017-09-27:
@chb0github see https://groups.google.com/forum/#!topic/gqrx/pYKfnykjHQ4

anroots, 2018-04-22:
Manually removing `librtlsdr.*` files in `/usr/local/lib`, then reinstalling as described by @ikizhvatov's link helped.

KC7NOA, 2018-06-15:
is it in a different folder in ubuntu 16.04?

thailemon, 2018-06-22:
i did the following on my RPI3:
sudo make uninstall in the rtl-sdr/ folder (from manual install)
sudo apt-get purge rtl-sdr
sudo apt-get -f install rtl-sdr
sudo ldconfig

shutterfire, 2019-01-06:
Any of these didn't help here.
DietPi is the distro.
I had everything working, the hassle started when I installed rpitx