gqrx-sdr/gqrx #557: include conflict : cmath vs math.h

jan2son, 2017-09-07:
I can't build sources (master branch, commit SHA 9648a14).
There is obviously include issue :

../qtgui/plotter.cpp: In member function ‘void CPlotter::drawOverlay()’:
../qtgui/plotter.cpp:1168:39: error: call of overloaded ‘abs(double)’ is ambiguous
     m_dBStepSize = abs(m_MaxdB-m_MindB)/(double)m_VerDivs;

It seems the qtgui/plotter.cpp includes some header whch includes with math.h . Maybe Makefile order could solve this issue.

I am using standard ubuntu 17.04, 
g++ -v
  gcc version 6.3.0 20170406 (Ubuntu 6.3.0-12ubuntu2)

make -v
  GNU Make 4.1
  Built for x86_64-pc-linux-gnu

csete, 2017-09-07:
> I can't build sources (master branch, commit SHA 9648a14).

That commit is from October 2014. Sorry but I have more than enough things to do already and don't see why we should be concerned about this.

jan2son, 2017-09-07:
Thank you! I will check latest commit.

jan2son, 2017-09-07:
I did use a wrong repository.

closed 2017-09-07 by jan2son