gqrx-sdr/gqrx #558: gqrx on ubuntu 17 seg fault

kf7rcs, 2017-09-12:
Zesty package crashes
$:~/source_code/gqrx/build$ apt list --installed | grep gqrx
gqrx-sdr/zesty,now 2.6-1 amd64 [installed]

$ gqrx
linux; GNU C++ version 6.2.0 20161027; Boost_106200; UHD_003.009.005-0-unknown

Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
FM demod gain: 1.52789
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/dogbert/.config/gqrx/bookmarks.csv
Segmentation fault (core dumped)
=======================================
so i try to build... and says no pulse audio, but pulse audio is installed...

cmake ../
  Could not find a package configuration file provided by "PulseAudio" with
  any of the following names:

    PulseAudioConfig.cmake
    pulseaudio-config.cmake
=========================================
:~/source_code/gqrx/build$ apt list --installed | grep pulse

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

gstreamer1.0-pulseaudio/zesty,now 1.10.4-1ubuntu1 amd64 [installed,automatic]
libcanberra-pulse/zesty,now 0.30-3ubuntu1 amd64 [installed,automatic]
libpulse-mainloop-glib0/zesty,now 1:10.0-1ubuntu2 amd64 [installed,automatic]
libpulse0/zesty,now 1:10.0-1ubuntu2 amd64 [installed,automatic]
libpulsedsp/zesty,now 1:10.0-1ubuntu2 amd64 [installed,automatic]
pulseaudio/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-dlna/zesty,zesty,now 0.5.2-1 all [installed]
pulseaudio-equalizer/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-esound-compat/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-bluetooth/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-droid/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-gconf/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-jack/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-lirc/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-raop/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-trust-store/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-module-zeroconf/zesty,now 1:10.0-1ubuntu2 amd64 [installed]
pulseaudio-utils/zesty,now 1:10.0-1ubuntu2 amd64 [installed]

csete, 2017-09-12:
Hi @kf7rcs 

This is a known problem with the packages in Ubuntu that has been discussed and and well documented, see e.g. issue #452 

One obvious workaround would be to use our PPA, which would also give you newer version of gqrx, see: http://gqrx.dk/download/install-ubuntu (please make sure to uninstall the earlier version before switching to the PPA!)

You are welcome to build from source, but it is a rather complicated process and it seems you don't know how to do that; I can't help you with that, sorry. Note that this is a bug tracker and not a user support forum, please see http://gqrx.dk/user-support if you need help.

Closed as duplicate of #452

closed 2017-09-12 by csete

kf7rcs, 2017-09-12:
Thanks for the reply, sorry for the dupe entry.

jkotrosa, 2018-01-16:
Had the same issue.  The solution was to install the pulse-audio client development headers and libraries

sudo apt-get install libpulse-dev

Then, everything went smoothly