gqrx-sdr/gqrx #559: UDP host setting does not persist

yuvadm, 2017-09-17:
When changing the UDP host setting from anything that isn't `localhost` the setting doesn't persist after program exit, it probably should.

csete, 2017-09-17:
It only happens with 127.0.0.1 - other IP addresses of host names work fine.

You can change the code here if you want:
https://github.com/csete/gqrx/blob/master/src/qtgui/dockaudio.cpp#L330

yuvadm, 2017-09-17:
Any reason it's excluded specifically?

csete, 2017-09-17:
Only that localhost and 127.0.0.1 are the same default and by principle we do not save the default value. I will change this but I have just shipped v2.8 yesterday so it will be a while before I will make a new release.

closed 2017-11-08 by csete

yuvadm, 2017-11-09:
@csete Thanks for the fix!

csete, 2017-11-09:
@yuvadm thanks for the feedback - I take it it solved your problem then :)

yuvadm, 2017-11-09:
Haven't tested yet, but looks like it should

commit a9bfa18d (2017-11-08): Always save UDP host string, unless it is empty

Fixes #559