gqrx-sdr/gqrx #560: new output from 'm' command on the TCP socket from gqrx 2.7 upwards

Marzona, 2017-09-19:
Hello,
since gqrx 2.7, getting the mode from TCP produces the mode and an additional string that was not present before.

on GQRX 2.6 I get:

<pre>
telnet localhost 7357
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
m
WFM_ST
</pre>

While on GQRX 2.7/2.8 I get:

<pre>
telnet localhost 7356
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
m
FM
10000
</pre>

Is this behavior expected?

(related to https://github.com/Marzona/rig-remote/issues/100)

csete, 2017-09-19:
Yes it is intended, see commit 38f06530b58a6cdf7976cb13cdaba4d6f7c6309a

Sorry for the inconvenience, but this appears to be the hamlib compatible way according to rigctld doc http://manpages.ubuntu.com/manpages/xenial/man8/rigctld.8.html

Marzona, 2017-09-20:
got it, thanks

closed 2017-09-20 by Marzona