gqrx-sdr/gqrx #561: ppm does nothing on CloudIQ

pidloop, 2017-10-09:
Hello,

I really enjoy using gqrx 2.8 on macOS 10.12 with my RFSpace CloudIQ, it works very very well. Thanks for this great program.

Just one issue please: the ppm correction field does not seem to do anything. I can change it to any value but nothing happens. Perhaps I am missing a step? Perhaps there is some communication with the CloudIQ that is not correct? I would be happy to help troubleshoot if I can do anything.

73, Elwood, WB0OEW

csete, 2017-10-09:
Hi Elwood,

It looks like it is not implemented by the backend:
https://github.com/csete/gr-osmosdr-gqrx/blob/master/lib/rfspace/rfspace_source_c.cc#L1570
So this problem exists for all RFSpace radios supported.

I'm curios, how much error do you have?

pidloop, 2017-10-09:
> rfspace/rfspace_source_c.cc#L1570

At 10 MHz WWV I measure 11.2 Hz too high. I can set this in the LNB LO but
that does not scale with frequency,

csete, 2017-10-09:
If / when I make a new release I will try to have this fixed as it shouldn't be difficult to fix.

pidloop, 2017-10-09:
Sounds great, no panic. Thank you.

costonisp, 2017-10-09:
I experience the same issue  with my PlutoSDR. The error is about 5ppm.

csete, 2017-10-09:
The same applies to plutosdr. In fact, the plutosdr backend is really just a prototype to get the receiver up and running and it is not even merged upstream. It will eventually require a maintainer, otherwise it will be a dead project. I simply don't have the time to maintain and package all these backends by myself.

csete, 2018-03-16:
I have now pushed changes to the [rfspace](https://github.com/csete/gr-osmosdr-gqrx/tree/rfspace) and [plutosdr](https://github.com/csete/gr-osmosdr-gqrx/tree/plutosdr) branch in [gr-osmosdr-gqrx](https://github.com/csete/gr-osmosdr-gqrx) that implement PPM setting for these devices. It will be available in the next packaging round in a few days.

pidloop, 2018-03-17:
Sweet! Many thanks Alex.

Elwood, WB0OEW

closed 2018-03-18 by csete