gqrx-sdr/gqrx #568: *** Error in `gqrx': corrupted double-linked list: 0x01818a00 ***

LudooduL, 2017-11-09:
Hello,
gqrx crashes when I start it, as shown below.

My configuration (a Pi Zero machine).
$ uname -a
Linux picam 4.9.59+ #1047 Sun Oct 29 11:47:10 GMT 2017 armv6l GNU/Linux

I have installed gqrx with apt-get install gqrx-sdr

Any idea?

Thanks, cheers,
Ludo.

-- Full execution trace:
$ gqrx -r
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

Controlport disabled
No user supplied config file. Using "default.conf"
Can not delete "/home/ludo/.config/gqrx/default.conf" - file does not exist!
Fontconfig warning: ignoring UTF-8: not a valid region tag
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
FM demod gain: 1.52789
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/ludo/.config/gqrx/bookmarks.csv
getDeviceList : Available input devices:
    0 : "RFSPACE SDR-IQ Receiver"
    1 : "RFSPACE SDR-IP Receiver"
    2 : "RFSPACE NetSDR Receiver"
    3 : "RFSPACE Cloud-IQ Receiver"
    4 : "RTL-SDR Spectrum Server"
    5 : "Red Pitaya Transceiver Server"
    6 : "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/ludo/.config/gqrx/default.conf"
*** Error in `gqrx': corrupted double-linked list: 0x00675640 ***
Aborted

csete, 2017-11-09:
Gqrx requires a multi core machine. Even the RPI3 is barely fast enough.

Please note that this is a bug tracker, not a user support forum, please see http://gqrx.dk/user-support

closed 2017-11-09 by csete

LudooduL, 2017-11-09:
Thanks for the information ... but I guess such an error is a bug, right?
I mean, gqrx could provide another information when this occurs?

Ludo.

csete, 2017-11-09:
Yes, it is a bug if it occurs on a supported platform using the latest released version (2.8), but distributions are often behind.

LudooduL, 2017-11-09:
You mean that the bug is not present in version 2.8?