gqrx-sdr/gqrx #570: AM produces no audio on RPI unless volk_profile has been run

csete, 2017-11-12:


gvanem, 2017-12-15:
Hello again GPredict guy.

I've also compiled GQRX using MSVC-2017, but I need some help with the audio. I didn't compile with 
`-DWITH_PORTAUDIO` due to this statement **pulseaudio or portaudio (Linux only and optional)** in the docs.

Wonder why since PortAudio is for Windows too!? So, I ass-u-med `gnuradio-audio.dll` would take care of
giving me sound. Not the case; all I hear is a crackling sound on a good FM-carrier. Spectrum seems 
okay (an audible sound in *SDR-Sharp*). What could be the reason for this?

csete, 2017-12-15:
Like I mentioned in the other thread, I never considered windows to be a target platform for gqrx and I am surprised that people spend time on it. Anyway, portaudio should work on any platform.

csete, 2017-12-15:
As I recall, the cmake files will fallback to gnuradio-audio by default on unhandled platforms. You can configure gnuradio-audio to use Portaudio, but that interface is of unknown status (likely broken) and that's why I wrote an embedded Portaudio interface. I think you can hack the top level cmake file to let you choose Portaudio on non-linux platform.

gvanem, 2017-12-15:
> ... I never considered windows to be a target platform for gqrx  ...

Sounds a bit fatalistic. IMHO, GQrx could be a very good candidate for the best SDR-program on Windows (one day rivaling SDR-Sharp).

> ... configure gnuradio-audio to use Portaudio.

I've done that and get sound in some samples. Like simple sinus tones etc. Not sure that's done correctly.

csete, 2017-12-16:
Nice of you to say that. Personally, I think gqrx is too broken at the architectural level to ever become such thing and I have been working on a replacement. Just keeping the lights on here until the new project is usable.

argilo, 2023-04-29:
I was able to reproduce this, and it appears the `volk_32fc_magnitude_32f` VOLK kernel is the culprit. The bug occurs when the `u_orc` version is selected, but not when the `generic` version is selected.

argilo, 2023-04-29:
When the input sample is `0`, the kernel produces an output of `NaN`. If the AM demodulator's DC remove feature is enabled, then even a single `NaN` is enough to permanently break the IIR block.

This behaviour occurs on Raspbian Buster (ARM) but seemingly not on Ubuntu 22.04 (x86_64).

argilo, 2023-04-29:
It looks like the problem comes from ORC. Its test suite indicates that `sqrtf(0)` returns `NaN` on ARM:

https://gitlab.freedesktop.org/gstreamer/orc/-/blob/72b4699314e0a6eeeb29cea33d0410612c80f533/orc-test/orctest.c#L587-591

argilo, 2023-04-29:
I've explained the root cause in https://github.com/gnuradio/volk/issues/622.

A possible workaround is to put the following in `~/.volk/volk_config`:

```
volk_32fc_magnitude_32f generic generic
```

This will prevent the broken VOLK kernel from being used.

argilo, 2023-12-06:
This was fixed in https://github.com/gnuradio/volk/pull/658, which made it into VOLK 3.1.0: https://github.com/gnuradio/volk/releases/tag/v3.1.0

The next major version of Raspberry Pi OS should pick up the fix, but sadly not until 2025. Until then, users can either apply one of the workarounds, or build VOLK, GNU Radio, and Gqrx from source.

closed 2023-12-06 by argilo