gqrx-sdr/gqrx #571: gqrx 2.9 - MacOSX segmentation fault: 11

vk4tec, 2017-11-13:
Running gqrx on Mac OSX 2.9 results in hit and miss results.

Most times it results in a segmentation fault: 11

Sometimes it works sometimes it does not

Runs get_devices 

Andrew

Quasar-ru, 2017-11-14:
I'm faced with same problem on MacOS. Rebuild gr-osmosdr without UHD supporting has solved problem.

csete, 2017-11-14:
@Quasar-ru It's why my bundle is distributed without UHD support.

Quasar-ru, 2017-11-15:
@csete I build it from sources, and by default UHD is on. Maybe this situation is same.

cycleuser, 2017-12-02:
@csete 

Your bundle, the Gqrx.app, still fails on macOS High Sierra 10.13.1 as usual.

csete, 2017-12-02:
@cycleuser It works for some, it does not work for others. It's why there is an open issue about it.

Like I said before, resolving this will require somebody to help me. My Macs are too old to run newer version of the OS and I can't afford buying new. Would be nice if people who are willing to throw shitloads of money at Apple would help.

cycleuser, 2017-12-03:
@csete

Like I did before, I just use VMware to run a macOS 10.13.1 under Windows. 
It performance OK comparing to my iMac 2013 which also runs 10.13.1.

My advice is that maybe you might try to run macOS as a virtual machine if there keeps the financial problem on getting an expensive brand new one.

csete, 2017-12-03:
Thanks for the advice. However, I would still prefer somebody else to look into this, as I have have plenty of other tasks to work on.

Quasar-ru, 2017-12-05:
I have rebuild gqrx from git repo on MacOS Sierra High (10.13.1) and with gnuradio 3.7.11. May be some one want to test it.

https://www.dropbox.com/s/mc2c0a76epv5i08/Gqrx.app.zip?dl=0

cycleuser, 2017-12-07:
@Quasar-ru  Awesome, man~ Let me try it ~

cycleuser, 2017-12-07:
Got the Dyld Error with QtNetwork"

```
Process:               gqrx [4209]
Path:                  /Applications/Gqrx.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.9 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [4209]
User ID:               501

Date/Time:             2017-12-07 13:12:05.136 +0800
OS Version:            Mac OS X 10.13.1 (17B1003)
Report Version:        12
Anonymous UUID:        EB4926E2-0E9B-B204-0102-1B480BE2D9B9

Time Awake Since Boot: 13000 seconds

System Integrity Protection: disabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtNetwork.framework/Versions/5/QtNetwork
  Referenced from: /Applications/Gqrx.app/Contents/MacOS/gqrx
  Reason: image not found

Binary Images:
       0x10a39f000 -        0x10a538ffb +dk.gqrx.www (2.9 - 0) <555C231B-3FBD-31F4-B843-EE2FE5108146> /Applications/Gqrx.app/Contents/MacOS/gqrx
       0x10da7d000 -        0x10dac798f  dyld (519.2.1) <CD85EFBA-B9D6-3FD5-B777-A23B2DDEF192> /usr/lib/dyld

Model: iMac14,2, BootROM IM142.0123.B00, 4 processors, Intel Core i7, 4 GHz, 32 GB, SMC 2.15f7
```

BH5HDJ, 2017-12-07:
@Quasar-ru Awesome, man~ Let me try it ~

BH5HDJ, 2017-12-07:
Process:               gqrx [33597]
Path:                  /Applications/Gqrx.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.9 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [33597]
User ID:               502

Date/Time:             2017-12-07 14:37:26.664 +0800
OS Version:            Mac OS X 10.13.1 (17B1003)
Report Version:        12
Anonymous UUID:        6F383320-6A8C-930F-77BF-89C7DE7BE074

Sleep/Wake UUID:       C3FBA0A9-F24E-4A50-8C38-8E2FEF537044

Time Awake Since Boot: 56000 seconds
Time Since Wake:       15000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtNetwork.framework/Versions/5/QtNetwork
  Referenced from: /Applications/Gqrx.app/Contents/MacOS/gqrx
  Reason: image not found

Binary Images:
       0x10a66e000 -        0x10a807ffb +dk.gqrx.www (2.9 - 0) <555C231B-3FBD-31F4-B843-EE2FE5108146> /Applications/Gqrx.app/Contents/MacOS/gqrx
       0x110575000 -        0x1105bf98f  dyld (519.2.1) <CD85EFBA-B9D6-3FD5-B777-A23B2DDEF192> /usr/lib/dyld

Model: MacBookPro10,1, BootROM MBP101.00F2.B00, 4 processors, Intel Core i7, 2.3 GHz, 16 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe, 1 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.190.15.1a5)
Bluetooth: Version 6.0.0f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: iPhone, Ethernet, en4
Serial ATA Device: APPLE SSD SM256E, 251 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 3.0 Bus
USB Device: iPhone
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4

Quasar-ru, 2017-12-07:
Yes, found problem with libraries in bundle. Now, bundle repacked. I have tested on clear (without development environment) Mac. Looks like now it workable :-)

https://www.dropbox.com/s/d39dshv9tphij3a/Gqrx_07.12.app.zip?dl=0

effeerre75, 2017-12-07:
Here Version 2.9-1 is working fine without any problem... MacBook Pro 2010 macOS 10.13.2. Thanks for the great works as always!

dl9rdz, 2017-12-07:
Here a freshly installed version 2.9 with an Airspy Mini connected kept on crashing with segmentation fault.

Running gqrx in gdb actually revealed the reason: 
> Using AirSpy MINI v1.0.0-rc9-0-ga56adfd 2016-06-12, samplerates: 3M 6M 
> ibc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Unsupported samplerate: 2.5M

So its easy to avoid this segfault by fixing the configuration, but it would be nice if gqrx would not silently crash with segfault just because of selecting an unsupported sampling rate, but handle the exception more smartly by printing our an error message

oe2wmi, 2017-12-08:
On a MacBook Pro  2010, macOS 10.13.2:  installation without problems, everything works fine

cycleuser, 2017-12-11:
Still not working on iMac14,2

> Process:               gqrx [3480]
Path:                  /Volumes/VOLUME/*/Gqrx.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.9 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [3480]
User ID:               501

Date/Time:             2017-12-11 09:54:49.535 +0800
OS Version:            Mac OS X 10.13.2 (17C88)
Report Version:        12
Anonymous UUID:        EB4926E2-0E9B-B204-0102-1B480BE2D9B9

Time Awake Since Boot: 2400 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff5119de3e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff512dc150 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff510fa312 abort + 127
3   libc++abi.dylib               	0x00007fff4f0d5f8f abort_message + 245
4   libc++abi.dylib               	0x00007fff4f0d6113 default_terminate_handler() + 241
5   libobjc.A.dylib               	0x00007fff50460eab _objc_terminate() + 105
6   libc++abi.dylib               	0x00007fff4f0f17c9 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff4f0f126d __cxa_throw + 121
8   libgnuradio-audio.3.7.10.1.dylib	0x000000010c15c6e4 gr::audio::osx_sink::setup() + 1060
9   libgnuradio-audio.3.7.10.1.dylib	0x000000010c15b7df gr::audio::osx_sink::osx_sink(int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) + 575
10  libgnuradio-audio.3.7.10.1.dylib	0x000000010c15b4a9 gr::audio::osx_sink_fcn(int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) + 57
11  libgnuradio-audio.3.7.10.1.dylib	0x000000010c13889e gr::audio::sink::make(int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool) + 1550
12  dk.gqrx.www                   	0x000000010b8b3826 receiver::receiver(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned int) + 3638
13  dk.gqrx.www                   	0x000000010b8a0884 MainWindow::MainWindow(QString, bool, QWidget*) + 1172
14  dk.gqrx.www                   	0x000000010b89b4db main + 5163
15  libdyld.dylib                 	0x00007fff5104e115 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff5119e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff512d926f _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff512d8c4d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff5119e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff512d926f _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff512d8c4d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff5119e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff512d926f _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff512d8c4d start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff5119e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff512d926f _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff512d8c4d start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff5119e562 __workq_kernreturn + 10
[... 339 lines cut ...]
===========                     =======  ======= 
Accelerate framework               128K        2 
Activity Tracing                   256K        2 
CG backing stores                  704K        5 
CoreImage                            8K        3 
CoreServices                        36K        2 
CoreUI image file                  332K        5 
Foundation                           4K        2 
Kernel Alloc Once                    8K        2 
MALLOC                           165.5M       45 
MALLOC guard page                   48K       11 
Memory Tag 242                      12K        2 
STACK GUARD                       56.0M       11 
Stack                             12.6M       11 
VM_ALLOCATE                       3304K       28 
__DATA                            33.2M      285 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__LINKEDIT                       197.4M       52 
__TEXT                           179.3M      280 
__UNICODE                          560K        2 
mapped file                       58.7M       23 
shared memory                     3360K       15 
===========                     =======  ======= 
TOTAL                            713.8M      770

KirilShishkoski, 2017-12-15:
If you getting segmentation fault: 11 just run the following command:

gqrx rebuild

I hope it solves your problem.

mkarliner, 2018-03-14:
Just tried the latest dropbox image and get 'no mountable file systems'

csete, 2018-03-18:
There is a new version 2.11.1 available using portaudio as audio backend. Hopefully it will work better.

closed 2018-03-18 by csete