gqrx-sdr/gqrx #574: GQRX in cronjob

c0debreaker, 2017-11-25:
Is it possible to record IQ in gqrx from a cronjob? I've given up on using rtl_fm since it's only the middle part of my NOAA APT weather image that is clean. The 25-30% of top and bottom are just noise when using rtl_fm even when I used `-s 60k` which is enough. I like to use GQRX instead.

csete, 2017-11-25:
We use github issues for bug tracking not as a support forum. Please see the website http://gqrx.dk/user-support

closed 2017-11-25 by csete

vk4tec, 2017-11-26:
Are you familiar with perl ?

You can all a satellite tracking program and say 

“is the elevation > 5 deg then see if a record job is started 

A

c0debreaker, 2017-12-27:
Sorry vk4tec, I just saw your post. I posted in the Google Groups with the same question I posted here but nobody responded to my question. And yes, I'm familiar with perl. What I was asking about was to how to use gqrx in command line similar to how rtl_fm is executed.

csete, 2017-12-27:
Sorry @c0debreaker, but I have got bored about discussing this topic over and over again. Anyway, here we go:

1. You can not run gqrx without a gui. I.e. there is no option to not show the GUI.

2. There are a few command line options (see -h) but not for setting everything you can set in the GUI. Such things are not necessary since gqrx remembers most of its settings; not all, but I think v2.9 now restores everything necessary to receive NOAA.

3. Once gqrx is running you can interact with it through the [remote control interface](http://gqrx.dk/doc/remote-control), in particular, you can use this to set frequency and to start/stop audio recording.

4. If you want an rtl_fm like command line utility, you should really look at something else. I'm pretty sure there are some gnu radio flowgraphs floating around; you are definitely not the first person to want to capture NOAA on a headless linux system.

5. I myself [have published something](http://oz9aec.net/radios/gnu-radio/noaa-weather-satellite-reception-with-gnu-radio-and-usrp) in the past that I'm sure can be modified to work with rtlsdr on recent systems. Note that GNU Radio Companion generates python files that you can execute without GUI!

Bottom line, use gqrx if you want a nice, interactive GUI, use something else if you want a headless receiver.

c0debreaker, 2018-01-06:
Thank you so much!