gqrx-sdr/gqrx #576: Intermediate frequency for panadapter [Feature]

ntoulasd, 2017-12-04:
I would like an input for Intermediate Frequency setting.
Example : rtl-sdr will listen all the time in IF frequency of my FT-857 68.33Mhz even if I change the display frequency to 7.020 Mhz.
All network commands must change display frequency, but not receiving frequency.

This is because I use hamlib command to set frequency to both my transceiver and my sdr.
https://github.com/ntoulasd/GQRX-Hamlib

csete, 2017-12-05:
Controlling the LNB_LO through the remote interface has been implemented for this specific purpose. Please see if you can do it using that.

Also, Simon's gqrx-hamlib does this synchronization I believe: https://github.com/g0fcu/gqrx-hamlib

ntoulasd, 2017-12-06:
LNB_LO is not working with IF(Intermediate Frequency). LNB_LO is for upconverter or downconderter making a shift in frequencies.
IF Frequency is a stable frequency all the time, not a shift.
Example if you receiving 7.020Mhz the IF is 68,33Mhz, if you receiving 14.080Mhz the IF is again 68,33Mhz.

This is what I want to do with GQRX, Hamlib, Ubuntu, rtl-sdr  and Yaesu FT-847d
https://squashpractice.wordpress.com/2016/01/11/rtl-sdr-pan-adapter-using-hdsdr-with-the-ic-751a/

Simons synchronization is the same as mine https://github.com/ntoulasd/GQRX-Hamlib
There is a separate antenna reception from rtl-sdr and tunes sdr and tranceiver in the same frequency. I want to use my Yaesu FT-857 front end reception and tapping in the IF to make a panadapter.

Thanks

csete, 2017-12-06:
LNB_LO just changes the display frequency.

With LN_LO at 0, tune to 68.33 MHz. You will be receiving 68.33 MHz.

Set LNB_LO to -61.31 MHz and the display will show 7.02 MHz but still receive 68.33 MHz
Set LNB_LO to -54.25 MHz and the display will show 14.08 MHz but still receive 68.33 MHz

Sorry, but I neither have time nor interest to implement anything more elaborate in gqrx. Like I have said on many occasions,. gqrx is a done deal from my point of view.

alexf91, 2017-12-06:
This should work:
https://gist.github.com/alexf91/6f229562a4ba59567e6189ca41c77915

I have used it when I tapped the IF of my FT817.

csete, 2017-12-06:
@alexf91 so, am I correct that it can be done with LNB_LO?

To be honest I am nnot 100% certain because I do not have experience with these scenarios, but I recall this to be the reason why we wanted LNB_LO exposed in the remote control interface.

alexf91, 2017-12-06:
@csete Yes, it can be done.
My script just reads the frequency from rigctld and sets LNB_LO to (frequency - IF).

ntoulasd, 2017-12-06:
Alex thank you. This is doing the job. 

LNB_LO = receiving_frequency-Intermediate_frequency

closed 2017-12-06 by ntoulasd