gqrx-sdr/gqrx #577: List of available configurations

gvanem, 2017-12-15:
It's rather confusing that GCrx doesn't list where the `.conf` file(s) are with the `-l, --list` option. 
Let alone the use of `$XDG_CONFIG_HOME` on Windows.

In my case it stores the `default.conf` in this irregular place: `C:/Users/Gisle/.config/gqrx/`.
But with this patch:
```diff
--- a/src/applications/gqrx/main.cpp 2017-12-14 13:53:07
+++ b/src/applications/gqrx/main.cpp 2017-12-15 11:22:47
@@ -240,7 +240,9 @@
         for (int i = 0; i < conf_files.count(); i++)
         {
             std::cout << "     ";
-            std::cout << conf_files.at(i).toLocal8Bit().constData() << std::endl;
+            std::cout << conf_path.toLocal8Bit().constData()
+                      << conf_files.at(i).toLocal8Bit().constData()
+                      << std::endl;
         }
     }
 }
```
it's a bit more clear where it is; `gqrx.exe -l` gives:
```
 Existing configuration files:
     C:/Users/Gisle/.config/gqrx/default.conf
```
Instead of the now brief and confusing output:
```
Existing configuration files:
    default.conf
```

csete, 2017-12-15:
As you point out, already the choice of configuration paths are broken on windows. I'm sorry, but windows support is really out of the scope for gqrx. In fact, it blows my mind that people bother using it when there are already several much better SDR applications for windows.

I can accept this change if it is put inside #ifdef windows but I don't want to change the behavior on linux and mac.

csete, 2018-03-16:
The output now includes the path:

```
gqrx -l

Existing configuration files in /home/alc/.config/gqrx/
     airspy-mini.conf
     airspy.conf
     airspyhf.conf
     airspymini.conf
     airspyverter-mini.conf
     airspyverter.conf
     cloudiq.conf
     ...
```

closed 2018-03-18 by csete