gqrx-sdr/gqrx #578: Crashes on startup when audio output device is gone

JonGretar, 2017-12-16:
I use virtual soundcards for routing audio output from `gqrx` in to `fldigi`. I usually use Jack2 these days. I only have Jack2 running when I am using `fldigi` so the virtual soundcard goes away. However if I forgot to deselect JackRouter from the audio output settings before turning off `gqrx` then it will crash on startup the next time. Quite understandably really. 

It's a minor issue as I just attempt to start it again and it will automatically reset these settings so the default audio will be selected but it will also reset things like window layout which is of a mild annoyance.

Probably just needs a check on startup to see if things like the selected audio output is still available and if not then just select the default audio output or pop up with the settings dialog.

I'm running macOS Sierra 10.12.6.

### Reproducing bug:

1. Start Jack2.
2. Start `gqrx`.
3. Select JackRouter for the audio output.
4. Stop `gqrx`.
5. Stop Jack2.
6. Attempt to start `gqrx`
7. Curse

JonGretar, 2017-12-16:
I'm guessing this would also happen if you were using things like external USB soundcards.

csete, 2017-12-16:
Unfortunately, gqrx is built on top of other frameworks which makes it extremely difficult, in some cases practically impossible, to properly handle such extreme cases.

The application settings are only stored when it is quit, but not if gqrx crashes. So I think that's why you loose the settings that you make during the session.

JonGretar, 2017-12-16:
> The application settings are only stored when it is quit, but not if gqrx crashes. So I think that's why you loose the settings that you make during the session.

Possibly a slight misunderstanding here. I'm not referring to gqrx crashing if I remove the sound card *while* gqrx is running. I think that would be quite understandable. 

I'm referring to gqrx being shut down correctly(The whole application itself, not DSP stopped). And while it's still shut down the sound card is removed. Then when the gqrx application is started again it crashes. I then start it again and I get some dialog asking if I wish to modify settings or something like that. No matter what I select there the window setup is reset to factory defaults.

However. If reading the configuration file on startup and setting the soundcard up to the pipeline is handled by some external library then, like I say, this is just a minor annoyance.

csete, 2017-12-16:
That scenario used to be handled pretty well, when gqrx is built to use the gnuradio-audio and gnuradio-audio is configured to us the native OS X audio backend. TBH, I think it even handles disappearing devices  while the program is running, but I don't remember.

JonGretar, 2018-03-17:
Works like a charm.
If I select the virtual soundcard as the output, close gqrx, delete the soundcard and then start gqrx again it just selects the default soundcard instead of crashing.

csete, 2018-03-17:
Cool, thanks for the feedback. I will stay with this Portaudio backend for now.

closed 2018-03-18 by csete