gqrx-sdr/gqrx #579: Everything works expect the spectrum

mehdideveloper, 2017-12-20:
Hi
I've compiled the latest Gqrx from git and it does not have a spectrum (although it has waterfall)
It was working before this, but this happened after I fetched and compiled the latest version.

Info:
Mac OS X 10.13.2
Gqrx v2.9-4-g4138038
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git)

Here's a screenshot:

[image]

Any ideas?
Thanks

alexf91, 2017-12-20:
> screen shot 2017-12-20 at 11 34 25 am

Start with `gqrx -r`.
The problem occurs sometimes after updates. The `-r` flag resets the 
configuration file and should fix the problem.

mehdideveloper, 2017-12-20:
Oh thank you very much.

closed 2017-12-20 by mehdideveloper