gqrx-sdr/gqrx #580: What happened to the AppImage?

probonopd, 2017-12-27:
What happened to the AppImage? It seems to be missing from the latest releases. Can you please add it back? Thanks.

csete, 2017-12-27:
I didn't have have any AppImage

probonopd, 2017-12-27:
Maybe you forgot ;-) https://github.com/csete/gqrx/releases/download/v2.6/Gqrx-2.6-1.glibc2.17-x86_64.AppImage

csete, 2017-12-27:
I mean, I didn't have any AppImage for v2.9. I remember trying to run the script I had from v2.6 but it didn't work and I didn't have time to investigate.

probonopd, 2017-12-27:
Should be straightforward these days using linuxdeployqt like in this example:
https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci

Let me know if you run into issues, I'm happy to help.

csete, 2017-12-27:
Ok, thanks, I will try it when I come back to work on gqrx again. For the time being I'm busy with other projects.

probonopd, 2017-12-27:
Looks like for whatever reason `linuxdeployqt` doesn't like dots in filenames before the suffix. Using names like `akira.desktop`, `akira.svg` and `usr/bin/akira` it is now working for me:

https://github.com/probonopd/Akira/blob/appimage/.travis.yml

If this build succeeds and if the resulting AppImage works, I will send a Pull Request if you like:
https://travis-ci.org/probonopd/Akira/builds/322340117

csete, 2017-12-28:
I think you meant to comment on another project?

probonopd, 2017-12-28:
Ouch, sorry. Multitasking is not easy ;-)

csete, 2018-03-26:
There is now AppImage of the latest release, thanks to a user who decided to investigate the problems.

@AsciiWolf  I would kindly ask you to not poke me for updates as I find them both meaningless and extremely annoying. I am putting a lot of additional effort into running the project openly where all relevant information is publicly available for anyone. Thanks for understanding.

closed 2018-03-26 by csete

probonopd, 2018-03-28:
Thanks @csete. I agree that "poke" type messages can be annoying. The following is not meant as a "poke" but really as a question:

How would you like AppImage generation as a part of the Travis CI script (if someone would send a Pull Request)? This way, it would not require additional work to produce the AppImage, as it would be an automatic by-product of each `git push`. And if a commit would break the AppImage generation, it would be immediately obvious which was the breaking commit.

csete, 2018-03-28:
@probonopd Now that I know how to build the AppImage again it doesn't take time. The time consuming part is packaging the dependencies for Ubuntu. THe AppImage recipe uses the Ubntu PPA so I'm not sure how that would work in Travis CI. We already have Travis CI builds of commits and pull requests.

probonopd, 2018-03-29:
We could [make the AppImage based on the Travis CI build artifacts](https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci) rather than using ingredients from the Ubuntu PPA. Let me know if you are interested and I can have a look at it.