gqrx-sdr/gqrx #582: cursor position +10MHz and up not possible

c5e3, 2018-01-01:
i can't move the cursor further than +10MHz from center:
[image]

moving it to the far left is possible though:
[image]

alexf91, 2018-01-01:
This behavior is caused by two bugs:
* Frequency Widget range in the Receiver Options dock is hardcoded to 7 digits
* Frequency Widget doesn't respect range limits when the number is negative

csete, 2018-01-01:
Sounds like a bug. We should probably get rid of the "filter offset" widget, perhaps keep it as a display widget, and implement free tuning with the main frequency controller.

If it is very annoying here and now, we can probably work around this by disconnecting the spectrum display from the filter offset and you can then click on the spectrum also outside the 9.999999 MHz.

csete, 2018-03-17:
I believe it is fixed now :)
[image]

closed 2018-03-18 by csete