gqrx-sdr/gqrx #583: crash on startup loading hackrf config

c5e3, 2018-01-01:
gqrx v2.9-4-g4138038

when using a HackRF with a new config file, it works perfectly. when i load a config with `-c hackrf.conf`, i get the following error at startup:
```Using HackRF One with firmware 2017.02.1 
terminate called after throwing an instance of 'std::runtime_error'
  what():  hackrf_set_freq(1e+12) has failed (-1000) Pipe error
Abgebrochen (Speicherabzug geschrieben)```

the error in german means `Aborted (Core Dumped)`

i don't get any error using gqrx 2.6 from the ubuntu repos

csete, 2018-01-01:
Presumably, you are also using different version of libhackrf and lignuradio-osmosdr when you are using v2.6 from Ubuntu repository. I can only re-iterate that gqrx does not do any hardware interfacing and there is usually very little chance that such errors are due to gqrx.

Also, please make sure to check the list of known issues before posting filing issues.

Closing as duplicate of issue #504.

closed 2018-01-01 by csete

c5e3, 2018-01-01:
i use `gr-osmosdr 823b9c22 (0.1.5git) gnuradio 3.7.10` and `libhackrf version: git-0335f1a (0.5)` with both, v2.6 and v2.9

csete, 2018-01-01:
You said gqrx v2.6 from the Ubuntu repository. That brings in its own dependencies. Are you overriding them?

csete, 2018-01-01:
Also, please provide your info on the other issue. It is very confusing to have info spread out over multiple issues.

c5e3, 2018-01-01:
ok, i will continue posting on the other issue