gqrx-sdr/gqrx #584: Mac OSX - I/Q files don't "play"

MStrecke, 2018-01-12:
When trying to "play" a recorded I/Q file in version 2.9.1 the "I/Q recorder" seems to play it (the slider moves). But when the DSP is started the signal is not being decoded and no spectrum or waterfall display is shown.
(See also the discussion under https://groups.google.com/forum/#!topic/gqrx/RJFLJGTsR1U ).

As I now found out that the reason for this was that the recorded file was not stored on the system partition.

When trying to replay the file from the (larger) data partition **on the same SSD drive**  no decoding is being performed. After copying it to the system partition everything works as expected.

When recording the file is stored on the selected folder (even on the data partition).

csete, 2018-01-12:
When you mentioned "other file system" I started to suspect that the error may occur if there are non-standard characters in the path. I mean characters that require escaping in the code, e.g. spacebar. I tried on Linux using a path containing a spacebar and I can now reproduce the error.

Can you confirm that it is the same for you?

PS: There is no Gqrx 2.9.1 - the latest release is 2.9, and the current git HEAD is 8 commits ahead of 2.9 so it will appear as v2.9-8-gb1336d1.

MStrecke, 2018-01-12:
Yes - removing spaces from volume and folder names mitigates the issue.

And you're right about the version number - sorry.

csete, 2018-01-12:
Ok, thanks. I have updated the title since it not a mac-only issue. Now the question is whether the problem is in gqrx or the GNU Radio file source block.

csete, 2018-03-17:
Hello @MStrecke 

I believe that I have fixed this problem with the above commit. The fix will be available in version 2.11 due in a few days. I hope you will be able to test and let me know that it works.

closed 2018-03-18 by csete